KiCad/kicad-packages3D #24: Should the models that are added here also be added to the kicad library repo?

poeschlr, 2017-03-19:
Should we (until this repo is used by kicad) add the 3d models also to the normal kicad library repo?

easyw, 2017-03-21:
I think we should, unless there is some automatic procedure to add all the PR of this repo to the correspondent folder in the kicad library repo...

SchrodingersGat, 2017-03-21:
I don't believe there is such a procedure.

I am (very busy but also) working on improving the 3D models downloading code in KiCad such that by v5 release we can remove the 3D shapes from the kicad-library repo.. Hopefully all goes to plan.

Do either of you know c++?

poeschlr, 2017-03-21:
> Do either of you know c++?

You really hope for a no here. Because everyone who knows it knows that it is a shitty language that got a lot of features added without thinking about what the core problems of c/c++ really are.

"C makes it easy to shoot yourself in the foot; C++ makes it harder, but when you do it blows your whole leg off." 
-  Bjarne Stroustrup

Now a serious answer:
I know object oriented programming. (Yes i know c++, but i don't believe that asking if one knows a language is the right question. One should ask if somebody knows a specific concept. Learning a language is a lot easier than learning a concept. There are lots of people who will tell you they know c++ but don't understand any concept of object oriented programming.)

Now the bad news: I don't have a lot of time until after shell eco marthon Europe (25.5.2017). After that i really need to finish university. (I have enough money to support myself until February next year. The good news is i only need my masters project and my masters thesis. Maybe also one exam if i failed the one i took last month.)
This means i have some limited time from 25.5. onward. (Or at least i hope that it will be a bit more than i have now.)

SchrodingersGat, 2017-03-21:
Sounds like you have about as much stuff going on as I do, then :) Good luck with the eco marathon!

poeschlr, 2017-03-22:
As described (by @SchrodingersGat) in pull request https://github.com/KiCad/packages3D/pull/26 this repo is not active yet. Currently files that are added here should also be added to the "old" library.
I will leave this issue open such that others can see this information aswell. (or should we put this info into the readme until this repo takes over?)

poeschlr, 2017-08-23:
For reference
https://github.com/KiCad/kicad-library/issues/1411

Shackmeister, 2017-12-02:
Since we are longer supporting the old repo I suppose we can close this? :)

closed 2017-12-02 by SchrodingersGat