KiCad/kicad-packages3D #88: missing models

pointhi, 2017-07-19:
For example: https://github.com/KiCad/kicad-library/pull/908

@jkriege2 are those models script generated? (looks like so) I think it would be a good idea to also add those models into this Repo.

Shackmeister, 2017-07-19:
I believe these are script generated by updating a spreadsheet inside Freecad and reexporting from there. Im not sure if they are scaled correctly (1/2.54) for WRL. Also a stepfile is missing, but that should be fixable. 
I think to add material properties, (not sure if that is considered a must for having it in the packages3D repo) it would be the easiest to port the script to cadquery which is still some work.

jkriege2, 2017-07-19:
@Shackmeister is correct ... also the models are NOT scaled correctly ;-)

I'm not so sure, how to proceeed with them. I didn't really succees in automatising the export process with StepUp (but also didn't put in too much work) ... my idea would ahve been to simply load (or regenerate ) the FCStd-files and use Steup to AUTOMATICALLY export the models ... but that didn't really work out.

Maybe we simply merge them over and wait until someone with more time/knowledge of StepUp redos them scripted, or finds an automated export?

JAN

Shackmeister, 2017-12-03:
shouldnt this be closed?
I believe alot of these have been reexported with stepup

closed 2017-12-03 by SchrodingersGat