KiCad/kicad-packages3D #104: FreeCAD release to be used for generating models

easyw, 2017-07-29:
Hi, 
since FC0.17 is still under heavy development in the STEP field and it is adding some almost hidden bugs,  I would suggest to use the latest FC0.16 6710 release version to create the 3D models with the scripts.

For ubuntu just use the right repo to obtain FC0.16 6710; 
for windows I have uploaded a release of FC0.16 6710 at this link:
[FreeCAD_0.16.6710-x64_dev_win_rc1](url=https://sourceforge.net/projects/kicadstepup/files/dev/FreeCAD_0.16.6710-x64_dev_win_rc1.7z/download)

hackscribble, 2017-07-29:
Thanks @easyw 

> For ubuntu just use the right repo to obtain FC0.16 6710;

The PPA for FreeCAD 0.16 stable is here: https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-stable

(It's now up to revision 6712).

easyw, 2017-08-01:
Latest update:
FC0.17 >=11707 is now fine when exporting STEP models...
recap:

### **FC0.15 4671** ✅
### **FC0.16 >=6710** ✅
### **FC0.17 >=11707** ✅

antoniovazquezblanco, 2018-07-12:
Maybe this information could be moved to the README file so that is available for everyone.

evanshultz, 2018-07-12:
@antoniovazquezblanco 
I'm hesitant to do that because versions change frequently and some other problem might creep up which renders a version above unsuitable in some situation. Our readme would likely rot.

Shackmeister, 2018-07-12:
Currently it should be fine to use what ever latest version is available. Bug is fixed in V0.15-V0.18 so its quite hard to get a version with the bug.. Personally I think this should be handled in the PR comments

easyw, 2018-07-12:
I think we should suggest FC stable 0.17 13522 as best option.
And CadQuery 1.2

antoniovazquezblanco, 2018-07-17:
Either way, if this changes so quickly I would suggest asking for FreeCAD version information in the PR and manually check, or if this does not change that much, either include it in the README or in the CONTRIBUTING files, but having this information here is not very visible...

poeschlr, 2018-07-17:
Freecad 0.17 is the current stable release. So it should not change very much. (Will only get bugfixes.)

nickoe, 2019-06-06:
Does it make sense to keep this but open? Should we mention the expected minimum version of freecad to use in the README?

Shackmeister, 2019-06-06:
Not really, pretty much any FC version you can get a hold of has the nescessary fixes

closed 2019-06-06 by Shackmeister