KiCad/kicad-packages3D #134: License change from GPL with symbol exception to CC-BY-SA 4.0

poeschlr, 2017-09-20:
Hello contributors. Hello @SchrodingersGat. 

The kicad developers have now decided on what license they want the libraries to use.
A change of license requires every contributor to agree to this change. (I also feel moraly bound to ask if it is ok to change the license.)

So without further useless talk: Is it ok for you guys that the license changes.

@Misca1234 is it possible that the contributions made by you are shown as made by @stoth? Is this a second accound of yours that shares an email? Or is this a different persons accound and github screws up somehow.

Similarly @Shackmeister it seems the contributions made by you are listed as made by @DASFrank 

As far as i can tell if @easyw, @jkriege2, @Misca1234, @Shackmeister, @SchrodingersGat, @patricklaf and @maqifrnswa agree we are golden here.

Shackmeister, 2017-09-20:
@poeschlr yes thats right
I here by agree that Kicad, Kicad Developers, Kicad Librarians can do what ever they want my contributions.
Frank Severinsen

hackscribble, 2017-09-20:
I have also contributed models. I am happy with the licence change.

poeschlr, 2017-09-20:
@hackscribble 
Did i miss you? Sorry. I'm really sorry. (Damn)

SchrodingersGat, 2017-09-20:
I am happy to have my contributions re-licensed

jkriege2, 2017-09-20:
Yes, all my contributions may be relicensed ...
Jan Krieger @jkriege2

jkriege2, 2017-09-20:
What about all the other contributors not listed above?

easyw, 2017-09-20:
@poeschlr @SchrodingersGat 
I would agree to change the license in a way to conserve the exception we added, to let users to be free to use the library data in their own projects without the obligation to share the project files under this or any other license agreement and giving the library a (minimal) protection in terms of copying.
At the same time I would suggest to ask to some lawyer at FSF which would be the best License to be applied. In my previous research I found gEDA exception the best, by I let you guys to go deeper.
Maurice

Shackmeister, 2017-09-20:
@easyw EFF was also suggested at the forum

stambaughw, 2017-09-20:
There are most likely contributors on the dev mailing list as well.
Before the libraries were split out from the source, anyone who wanted
to contribute to the libraries did so through the dev list.

I seem to remember making some library changes before the libraries were
split out from the source but I could be wrong.  If I did make any
library commits, I give my permission to change the license.

stambaughw, 2017-09-20:
I asked the EFF over two years ago and pinged them at least 4 or 5
times.  I never received a response.  I have gave up on the EFF and gave
the CERN legal department a go.  They drafted the exception clause.

poeschlr, 2017-09-20:
Just to clear it up. This is a thread for the new 3d lib only. This is also why there are so few contributors listed.
It is ok if people who did not contribute here also give their consent but for other libs we will likely go a different route. (Only the new libs for v5 will get the new license. The "legacy" lib = current lib will stay at the current license.)

easyw, 2017-09-20:
@poeschlr @SchrodingersGat @stambaughw 
I would then agree to adhere to the CERN legal department suggestion... This is in fact very close to our 3D packages license model and already perfectly aligned with our main intent. Luckily we have started this 3D library with a clear statement since the beginning.

poeschlr, 2017-09-20:
There is also one question remaining.
Our models have the old license as comment in them. Should we re-build them?

Shackmeister, 2017-09-20:
@poeschlr wouldnt a search and replace be possible? both STEP and VRML are human readable. @easyw can you add the new licensing in Stepup and the CQ generator? :)

poeschlr, 2017-09-20:
It is only the script generated models. If we only update the models them selves the old license will resurface if we ever need to change something in the models. Might be best to change the license in the generator scripts, re generate and upload again. (I don't think a search and replace would be less work.)

easyw, 2017-09-20:
+1 for a search& replace recursive script 
That could do the massive job automatically

easyw, 2017-09-20:
Regenerate all bunch of models would also be error prone

Shackmeister, 2017-09-20:
Well Im planning to go through the all different families (resistors, QFN, QFP, SOIC etc) and make sure all of them have a model, in that case Im probably gonna remake quite a few (lot of stuff has changed since then). Unfortunately I don't know when I will have the time

easyw, 2017-09-20:
I would wait for:
1) a licence text that has been approved
2) then I would make an one shot massive replace text inside the text of the models of the repo
3) finally I would update gradually all the scripts to add the right licence, as soon as there is the need of new ones

patricklaf, 2017-09-21:
I agree to have my contributions re-licensed.

Misca1234, 2017-09-22:
As the physical person behind Misca1234 and stolth
I here by agree that Kicad, Kicad Developers, Kicad Librarians can do what ever they want my contributions.
Stefan Olsson

wiebus, 2017-09-22:
Hello Mr. Pöschl.

I agree with CC-BY-SA 4.0 as suggested.

But as far as i remember i published my library contributions as public
domain or cc0, so you would not need my agree.

Perhaps one day i will be able again to contribute moore.

With best regards: Bernd Wiebus alias dl1eic

Am Wed, 20 Sep 2017 09:25:54 +0000 (UTC)
schrieb Rene Pöschl <notifications@github.com>:

> Hello contributors. Hello @SchrodingersGat.
>
> The kicad developers have now decided on what license they want the
> libraries to use. A change of license requires every contributor to
> agree to this [...]

-- 
Dipl. Ing. Bernd Wiebus

Am Sportfeld 4
51399 Burscheid
Germany

Tel. 02174-7838210 (fest)
     0162-6157950  (mob)

bernd.wiebus@gmx.de

poeschlr, 2017-09-25:
@Misca1234: for the packages 3d repo we are double save. We have the github history plus the credits file that lists who did what.

And if your models are script generated than you hopefully included your name in the license of the generated models. (This is added to the files themselves.)

@wiebus: i think you did not yet contribute to this part of the kicad lib. (This thread is only about the new packages3d repo) For symbol and footprint libs we have a separate thread: https://github.com/KiCad/kicad-library/issues/1645

wiebus, 2017-09-26:
Hello Mr. Pöschl

> i think you did not yet contribute to this part of the kicad
> lib.

I contributed a long time ago. As an example the Mesh models for the
LEM-HallEffect Current Transducers, the old sovjet Area Transistors,
the Tyco BNC Socket or the IR-directFET SMD Housings. ;O)

Those Models are somewhere in
https://github.com/KiCad/kicad-library/tree/master/modules and of
course, they may be altered now, because i created them all in mm scale.

With best regards: Bernd Wiebus alias dl1eic

Am Mon, 25 Sep 2017 23:21:01 +0000 (UTC)
schrieb Rene Pöschl <notifications@github.com>:

> @Misca1234: for the packages 3d repo we are double save. We have the
> github history plus the credits file that lists who did what.
>
> And if your models are script generated than you hopefully [...]

-- 
Dipl. Ing. Bernd Wiebus

Am Sportfeld 4
51399 Burscheid
Germany

Tel. 02174-7838210 (fest)
     0162-6157950  (mob)

bernd.wiebus@gmx.de

poeschlr, 2017-09-26:
@wiebus sorry for the confusion.
I do not wish to say you did not contribute to the kicad library nor do i wish to say you did not contribute to the 3d model lib. But according to the git history of this new repo you did not (yet) contribute. 

The new lib (this repo) focuses on step compatible models. No contributions lacking a step model will be moved here. (They will still stay at the old repo which does not change it's license.) 

If your contributions are made using any CAD tool capable of exporting to step we can include them here. (This would not require re licensing because you would commit something new to a lib that already has the new license.) 
If they are made using wings we sadly can not use them in this repo. (Even if they are in mm scale.)

wiebus, 2017-09-26:
Hello Mr. Pöschl.

> I do not wish to say you did not contribute to the kicad library nor
> do i wish to say you did not contribute to the 3d model lib. But
> according to the git history of this new repo you did not (yet)
> contribute. 
> 

Ah, ok, i see.

> If they are made using wings we sadly
> can not use them in this repo. (Even if they are in mm scale.)

Why? I just imported a wrl file into FreeCAD, modified it, and exported
it to step. Look at the attachment
"NF-Transformer_ETAL_P1165_Faktor03937_RevA_30Aug2012.wrl" (Original)
and the convertet
"NF-Transformer_ETAL_P1165_Faktor03937_RevA_30Aug2012.step"

Is this step file ok for the library?

If so, i will convert some of the wings/wrl files i made.

With best regards: Bernd Wiebus alias dl1eic

Am Tue, 26 Sep 2017 10:37:12 +0000 (UTC)
schrieb Rene Pöschl <notifications@github.com>:

> @wiebus sorry for the confusion.
> I do not wish to say you did not contribute to the kicad library nor
> do i wish to say you did not contribute to the 3d model lib. But
> according to the git history of [...]

-- 
Dipl. Ing. Bernd Wiebus

Am Sportfeld 4
51399 Burscheid
Germany

Tel. 02174-7838210 (fest)
     0162-6157950  (mob)

bernd.wiebus@gmx.de

Content-Transfer-Encoding: base64
Content-Disposition: attachment;
 filename=NF-Transformer_ETAL_P1165_Faktor03937_RevA_30Aug2012.wrl

I1ZSTUwgVjIuMCB1dGY4CiNFeHBvcnRlZCBmcm9tIFdpbmdzIDNEIDEuNS4zCkRFRiBib2R5IFRy
YW5zZm9ybSB7CiAgY2hpbGRyZW4gWwogICAgU2hhcGUgewogICAgICBhcHBlYXJhbmNlIEFwcGVh
cmFuY2UgewogICAgICAgIG1hdGVyaWFsIERFRiBib2R5IE1hdGVyaWFsIHsKICAgICAgICAgIGRp
ZmZ1c2VDb2xvciAwLjAxOTYwOCAwLjM0NTA5ODAzOTIxNTY4NjI3IDEuMAogICAgICAgICAgZW1p
c3NpdmVDb2xvciAwLjA1MzMzMzMzMzMzMzMzMzM0IDAuMzA2NjY2NjY2NjY2NjY2NjQgMS4wCiAg
ICAgICAgICBzcGVjdWxhckNvbG9yIDAuMDk1NTg4MjM1Mjk0MTE3NjQgMC4yNzY0NzA1ODgyMzUy
OTQxMyAxLjAKICAgICAgICAgIGFtYmllbnRJbnRlbnNpdHkgMC40NTU1NTU1NTU1NTU1NTU1NQog
ICAgICAgICAgdHJhbnNwYXJlbmN5IDAuMAogICAgICAgICAgc2hpbmluZXNzIDEuMAogICAgICAg
IH0KICAgICAgfQogICAgICBnZW9tZXRyeSBJbmRleGVkRmFjZVNldCB7CiAgICAgICAgbm9ybWFs
UGVyVmVydGV4IFRSVUUKICAgICAgICBjb2xvclBlclZlcnRleCBUUlVFCiAgICAgICAgY29vcmQg
Q29vcmRpbmF0ZSB7IHBvaW50IFsKICAgICAgICAgIC03LjYgLTkuNzUgMTEuMCwKICAgICAgICAg
IC03LjYgOS43NSAxMS4wLAogICAgICAgICAgNy42IDkuNzUgMTEuMCwKICAgICAgICAgIDcuNiAt
OS43NSAxMS4wLAogICAgICAgICAgLTcuNiAtOS43NSA4Ljg4MTc4NDE5NzAwMTI1MmUtMTYsCiAg
ICAgICAgICAtNy42IDkuNzUgOC44ODE3ODQxOTcwMDEyNTJlLTE2LAogICAgICAgICAgNy42IDku
NzUgOC44ODE3ODQxOTcwMDEyNTJlLTE2LAogICAgICAgICAgNy42IC05Ljc1IDguODgxNzg0MTk3
MDAxMjUyZS0xNiwKICAgICAgICAgIC02LjA4IC05Ljc1IDExLjAsCiAgICAgICAgICAtNC41NjAw
MDAwMDAwMDAwMDA1IC05Ljc1IDExLjAsCiAgICAgICAgICAtMy4wNDAwMDAwMDAwMDAwMDA1IC05
Ljc1IDExLjAsCiAgICAgICAgICAtMS41MjAwMDAwMDAwMDAwMDA1IC05Ljc1IDExLjAsCiAgICAg
ICAgICAtNC40NDA4OTIwOTg1MDA2MjZlLTE2IC05Ljc1IDExLjAsCiAgICAgICAgICAxLjUxOTk5
OTk5OTk5OTk5OTYgLTkuNzUgMTEuMCwKICAgICAgICAgIDMuMDM5OTk5OTk5OTk5OTk5NiAtOS43
NSAxMS4wLAogICAgICAgICAgNC41NiAtOS43NSAxMS4wLAogICAgICAgICAgNi4wOCAtOS43NSAx
[... 780 lines cut ...]
LjAsCiAgICAgICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgMC45NzUwMDAwMDAwMDAwMDE0IDExLjAs
CiAgICAgICAgICAwLjc1OTk5OTk5OTk5OTk5OTcgMC45NzUwMDAwMDAwMDAwMDE0IDExLjAsCiAg
ICAgICAgICAwLjM3OTk5OTk5OTk5OTk5OTYgMC45NzUwMDAwMDAwMDAwMDE0IDExLjAsCiAgICAg
ICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgMS4zMzIyNjc2Mjk1NTAxODc4ZS0xNSAxMS4wLAogICAg
ICAgICAgMC43NTk5OTk5OTk5OTk5OTk3IDEuMzMyMjY3NjI5NTUwMTg3OGUtMTUgMTEuMCwKICAg
ICAgICAgIDAuMzc5OTk5OTk5OTk5OTk5NiAxLjMzMjI2NzYyOTU1MDE4NzhlLTE1IDExLjAsCiAg
ICAgICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgLTAuOTc0OTk5OTk5OTk5OTk4OCAxMS4wLAogICAg
ICAgICAgMC43NTk5OTk5OTk5OTk5OTk3IC0wLjk3NDk5OTk5OTk5OTk5ODggMTEuMCwKICAgICAg
ICAgIDAuMzc5OTk5OTk5OTk5OTk5NiAtMC45NzQ5OTk5OTk5OTk5OTg4IDExLjAsCiAgICAgICAg
ICAxLjEzOTk5OTk5OTk5OTk5OTcgLTEuOTQ5OTk5OTk5OTk5OTk4OCAxMS4wLAogICAgICAgICAg
MC43NTk5OTk5OTk5OTk5OTk3IC0xLjk0OTk5OTk5OTk5OTk5ODggMTEuMCwKICAgICAgICAgIDAu
Mzc5OTk5OTk5OTk5OTk5NiAtMS45NDk5OTk5OTk5OTk5OTg4IDExLjAsCiAgICAgICAgICAxLjEz
OTk5OTk5OTk5OTk5OTcgLTIuOTI0OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjc1OTk5
OTk5OTk5OTk5OTcgLTIuOTI0OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjM3OTk5OTk5
OTk5OTk5OTYgLTIuOTI0OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAxLjEzOTk5OTk5OTk5
OTk5OTcgLTMuODk5OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjc1OTk5OTk5OTk5OTk5
OTcgLTMuODk5OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjM3OTk5OTk5OTk5OTk5OTYg
LTMuODk5OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgLTQu
ODc0OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjc1OTk5OTk5OTk5OTk5OTcgLTQuODc0
OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjM3OTk5OTk5OTk5OTk5OTYgLTQuODc0OTk5
OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgLTUuODUgMTEuMCwK
ICAgICAgICAgIDAuNzU5OTk5OTk5OTk5OTk5NyAtNS44NSAxMS4wLAogICAgICAgICAgMC4zNzk5
OTk5OTk5OTk5OTk2IC01Ljg1IDExLjAsCiAgICAgICAgICAxLjEzOTk5OTk5OTk5OTk5OTcgLTYu
ODI0OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjc1OTk5OTk5OTk5OTk5OTcgLTYuODI0
OTk5OTk5OTk5OTk5IDExLjAsCiAgICAgICAgICAwLjM3OTk5OTk5OTk5OTk5OTYgLTYuOD

poeschlr, 2017-09-26:
It seems github removes attachments from mails.

I just tried to replicate your suggested workflow. Sadly it produces an empty step file with the error message (in the report view) `'wrl_test' is not a shape, export will be ignored.`
I use a quite old 0.17 nightly build version on linux. Maybe you have a newer version?
Did you try to reimport the generated step file?

Shackmeister, 2017-09-26:
@wiebus don't bother with the directfets, I have made a script for the models, but Im still reworking the footprints before adding the new models https://user-images.githubusercontent.com/5279723/29998978-85f2b142-9038-11e7-914b-7be0b0780de7.png

wiebus, 2017-09-26:
Hello Mr. Pöschl.

> It seems github removes attachments from mails.

This is sad. Can you give me your direct E-Mail? So i can send you 
My adress is bernd.wiebus@gmx.de

> I just tried to replicate your suggested workflow. Sadly it produces
> an empty step file with the error message (in the report view)
> `'wrl_test' is not a shape, export will be ignored.` I use a quite [...]

It is a bit tricky i think. ;O)

I opened my wings file again in Wings3d, and exported it as "collada"
file (dae).
This dae file was again opened in FreeCad.
There i had to reassign colour and "upgrade" the "meshes" to
"shapes"
This is wat the error message means.
At last, when i have only shapes (FreeCad named them "shells" at the
"combo" (inspector) window), i can mark them alltogether and export them
into a step file.

But there are drawbacks: The meshes have to be clean. But there might
be some selfintersections or similar oddities. In this case you will
get an error message like "unable to upgrade these objects" At an
example this is so with the body of the bnc socket, sadly.

> Maybe you have a newer
> version?

OS: Debian GNU/Linux 8.9 (jessie)
Word size: 32-bit
Version: 0.14.3702 (Git)
Branch: releases/FreeCAD-0-14
Hash: b3368125c63289ec8ce9faec2b2ae4c78d436406
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
SoQt version: 1.6.0a
OCC version: 6.7.0

> Did you try to reimport the generated step file?

Yes. It worked.

With best regards: Bernd Wiebus alias dl1eic

Am Tue, 26 Sep 2017 08:06:46 -0700
schrieb Rene Pöschl <notifications@github.com>:

> It seems github removes attachments from mails.
>
> I just tried to replicate your suggested workflow. Sadly it produces
> an empty step file with the error message (in the report view)
> `'wrl_test' is [...]

-- 
Dipl. Ing. Bernd Wiebus

Am Sportfeld 4
51399 Burscheid
Germany

Tel. 02174-7838210 (fest)
     0162-6157950  (mob)

bernd.wiebus@gmx.de

easyw, 2017-09-26:
Hi @wiebus 
the method you describe is the one I suggested many years ago, before there were a good support for STEP in FreeCAD.
https://youtu.be/qSkWBwRJVqg
Unfortunately this method gives you just a tessellate approximation of the real model... wrl are tessellated models made by triangles, so no curves are really present in the model... a cylinder will be a faceted tube, the same for a simple hole... Dimensions will be just approximated as the mesh will...
This is not the best way to get a mechanical integration between ECAD and MCAD.
For that reason then I made StepUp... the right path for modelling part is in a real mechanical modeler as FreeCAD or other mechanical sw... from that model it is possible to export a wrl approximation, useful for rendering purposes ... in many cases the rendering result is much better with wrl than with STEP... but in term of mechanical integration the need to have real model is a must.
For that reason in this new KiCad package 3D library, the models need to come from a mechanical modeler, best FreeCAD as it is open source and powerful.
Thank you for your contribution and Best Regards,
Maurice

wiebus, 2017-09-26:
Hello Maurice.

> Unfortunately this method gives you just
> a tessellate approximation of the real model... wrl are tessellated
> models made by triangles, so no curves are really present in the
> model... a cylinder will be a faceted tube,

Yes, i noticed this, but did not bother.....

> Dimensions will be just approximated as the mesh will... This
> is not the best way to get a mechanical integration between ECAD and
> MCAD.

Ah, i see.

> ... in many cases the
> rendering result is much better with wrl than with STEP... but in
> term of mechanical integration the need to have real model is a must.
> For that reason in this new KiCad package 3D library, the models need
> to come from a mechanical modeler,

Ok, this is a reason i undrerstand.

But a question: For this reason, you have to keep two models in the
library. WRL for rendering, and the step modell for export, where
precision matters. Do i see this right?

With best regards: Bernd Wiebus alias dl1eic

Am Tue, 26 Sep 2017 12:19:06 -0700
schrieb easyw <notifications@github.com>:

> Hi @wiebus
> the method you describe is the one I suggested many years ago, before
> there were a good support for STEP in FreeCAD.
> https://youtu.be/qSkWBwRJVqg Unfortunately this method gives you just [...]

-- 
Dipl. Ing. Bernd Wiebus

Am Sportfeld 4
51399 Burscheid
Germany

Tel. 02174-7838210 (fest)
     0162-6157950  (mob)

bernd.wiebus@gmx.de

easyw, 2017-09-26:
> But a question: For this reason, you have to keep two models in the library. WRL for rendering, and the step modell for export, where precision matters. Do i see this right?
Yes ...
I started with two kind of models in library because kicad was not able to render STEP models, so I made STEP source model and from it exported a WRL counterpart to be used in kicad.
Then I discovered that in kicad wrl could have more properties in terms of light and other  related to rendering...
So I improved the wrl exporter to get the best of both worlds :smile:
Now kicad in dev flavour is supporting natively step, but I still prefer to work in a mixed mode:
WRL to have nicer rendering plus transparency (not supported by step models) in kicad;
STEP model in FreeCAD where it useful for mechanical design.
The library has grown in this direction, giving a STEP support for users of kicad stable and WRL nice rendering for user's of both stable and development release...
You may find useful my FC StepUp WB
https://forum.kicad.info/t/kicad-stepup-the-sketcher-for-getting-to-blinky/7826
BR 😄 
Maurice

evanshultz, 2018-07-12:
Yes, I should think so.

antoniovazquezblanco, 2018-07-17:
@poeschlr you opened this. Should we close it? Thanks

closed 2018-07-17 by poeschlr