KiCad/kicad-packages3D #145: DIP sockets confusion

terjeio, 2017-10-20:
I have made a FreeCAD/CadQuery script for generating turned pin DIP-socket models but are somewhat confused of how to contribute since there are currently two folders for them. One is "Sockets_DIP.3dshapes" - this contains 3d models for the chip housings not sockets - all obsolete? The second is "Housings_DIP.3dshapes", the corresponding footprints already has a model filename added but no corresponding model file. Which is the correct one to update?

I have used drawings from TE Connectivity as a rough basis for the models, there are many variants available so should the maufacturer name be added to the model name? Or should I just keep the models for myself? They add up to 40MB+ (FCStd, WRL and step files), this likely because I am a beginner in 3D design and FreeCAD makes my brain hurt...

[image]

Attached is my script as it is currently - it needs some tuning but I have got the one socket I wanted so I do not want to spend more time until I know if the models are of use.

[main_generator.zip](https://github.com/KiCad/kicad-packages3D/files/1400927/main_generator.zip)

Finally a thought crossing my mind, should only the scripts be published for some of the more esoteric models so that end users must generate them themselves?

easyw, 2017-10-20:
The FC CadQuery scripts are hosted here:
https://github.com/easyw/kicad-3d-models-in-freecad
as in the readme

easyw, 2017-10-21:
@Shackmeister 
which is the right repo for pin DIP-socket models?
look at this model... it seems there is a new MCAD hero here ready to join the squad! 😸

jkriege2, 2017-10-21:
since the footprints reise here https://github.com/KiCad/Housings_DIP.pretty ...  the 3D models shall be in `Housings_DIP.3dshapes`

easyw, 2017-10-21:
> I have used drawings from TE Connectivity as a rough basis for the models, there are many variants available so should the manufacturer name be added to the model name?

I think what you have done is more than enough for the socket family! They look great and fit on footprint perfectly 😄 
PS if you want to have a check of the fp vs 3d model, you can add some lines to the code to load the fp, when generating the models (have a look at BGA models script)

> Finally a thought crossing my mind, should only the scripts be published for some of the more esoteric models so that end users must generate them themselves?

The best is merging all the generated models to the 3D package library and the scripts to the 3D script repo. The majority of users don't want to build anything.. they just prefere to have models ready to go.

Thanks for your contribution!
Maurice

terjeio, 2017-10-21:
ok, I will be on the move for the next few days so it may be a little while before I upload.

closed 2017-11-03 by terjeio