KiCad/kicad-packages3D #156: Simultaneous support for v4 and v5

fauxpark, 2017-11-11:
I'm wondering how support for 3D models in KiCad 4 will be accomplished after v5 hits. There is a new monolithic footprints repo that it will use instead, and the individual repos used by v4 have been frozen. But, this repository appears to be planned for use in v5 as well.

The new KLC defines a new naming convention for many footprints, so the filenames between the old and new footprints may not be the same. As the filenames for the 3D models are required to match the footprints, they will have to be renamed here as well. But then, many of them will no longer show up in v4!

Admittedly, 3D models are more of a "nice thing to have" rather than something more critical like the footprints themselves, but still, they are a nice thing to have, and I'm sure there are many people out there who won't/can't upgrade for whatever reason.

SchrodingersGat, 2017-11-11:
Good question :) this has already been "solved" as follows:

This 3d repo mirrors the 3d models in the old kicad-library repository. The individual .pretty repos can use the models stored in the old repo. 

Now that we are transitioning to v5 libs, the models in this repo should be renamed to match the new conventions in the kicad-footprints repo. 

Does this make sense?

fauxpark, 2017-11-11:
Ah, thanks. I don't know how far this repo has overtaken the old one but I don't think I'll mourn the loss. Apart from the STEP stuff the models are, again, not required for producing boards :P

poeschlr, 2017-11-11:
The old repos have more models than this one. All contributions (until now) have been submitted to both repos.

closed 2017-11-18 by fauxpark