KiCad/kicad-packages3D #157: Pin Header 3D Models

SchrodingersGat, 2017-11-20:
@DASFrank @Shackmeister @easyw 

The [new footprints repository](https://github.com/kicad/kicad-footprints) is well under way and we have now split this repository off from the legacy 3D models in the kicad-library repo.

To this end, the amazing models that you 3D wizards have scripted need to be re-jigged to match the new directory and naming structure.

The first example I can think of is the PinHeaders. Two major changes: 

1. They are now split into separate directories based on pitch
1. The naming is now `_Px.xxmm` instead of `_Pitchx.xxmm`

Can you please rebuild these models if that needs to be done, or simple shuffle them around and rename them :)

Our new libs are going to be great! The 3D models are especially awesome!

poeschlr, 2017-11-20:
Not only the pin header models need to be re organized. I think all models will need tweaking. (Once my connector footprint lib overhaul is finished i will re export the 3d models for my phoenix, molex and jst connectors.)

easyw, 2017-11-20:
@SchrodingersGat @poeschlr 
there still the License issue inside the models to be aligned to the CC new one from GPL

poeschlr, 2017-11-20:
That's why i will re export my models instead of simply renaming them.

easyw, 2017-11-20:
For the entire repo this would be a HUGE rework, unless done by a find/replace macro...
For the License @poeschlr please share the template text you are going to use, so who is going to manage this issue will have a canvas to use/replicate.
@SchrodingersGat  also a list of what has to be changed in the 3D repo because of the new footprints repo would help the maintainers

poeschlr, 2017-11-20:
A list of changes might also be necessary for reworking symbol links

Shackmeister, 2017-11-20:
Ill be happy to fix all my scripted models along with the KLC 3.0 footprints are being merged (or at least the naming has been updated). Im currently trying to make a 3D script compatible with @poeschlr new RLC script. Problem Im facing ATM is they share sizes between types, which doesnt workout for pin band and height :/ Im probably just gonna redo the parameters inside the old 3D scripts

 @easyw will the new license be added in the script so it will be automatically put in the models?

easyw, 2017-11-20:
> @easyw will the new license be added in the script so it will be automatically put in the models?
that is the plan 😄 
We need to have a template, then it can be just added ... I think I would need to ask contributors to opt for the change of the license for the created models from GPL with the exception to CC with the same exception...
but it could be redundant because they already moved to CC in the new package3d repo...

Shackmeister, 2017-11-22:
@SchrodingersGat @poeschlr @easyw
Who will be making the license template? Is it already made?

SchrodingersGat, 2017-11-22:
http://kicad-pcb.org/libraries/license/

nickoe, 2018-01-07:
Are there still open ends for v5 release?

Shackmeister, 2018-01-07:
Yes plenty

evanshultz, 2018-07-16:
How about now?

Shackmeister, 2018-07-16:
Yup still some, but won't be able to fix all of them :)

closed 2019-04-06 by Shackmeister