KiCad/kicad-packages3D #164: 3D models for Espressif modules

sandhan26, 2017-11-26:
Hey, guys! I have started working on designing models, footprints and symbols for Espressif modules, which are used in most of the IOT projects today.

While designing, the models, i face some challenges, which i need help with.
1. Incomplete data on dimensions in datasheet.
2. Mapping graphics on top of a face in FreeCAD.
3. How to properly extract dimensions data from Gerber files? 

For incomplete data, the datasheets provide at least the pad information and boundary dimensions. 
The rest of the details, which are needed for detailed 3D modelling are missing.

 [image]

is there any way to put the above graphics and text onto my FreeCAD model?

[image]

Some of the designs like antenna and holes on the pads had to be taken from Gerber files, by converting them to kicad_pcb and then measuring the dimensions using distance tool.

But, i think there might be a much easier way to do the same.

Above model is [ESP-WROOM-02](http://espressif.com/sites/default/files/documentation/0c-esp-wroom-02_datasheet_en.pdf)

easyw, 2017-11-26:
Hi @sandhan26 
nice model :smile:
> is there any way to put the above graphics and text onto my FreeCAD model?

for text you can use [Draft Shapestring](https://www.freecadweb.org/wiki/Draft_ShapeString)...
but in general Text will increase the size of the 3D model a lot... so, as suggested in the [README](https://github.com/KiCad/kicad-packages3D/blob/master/README.md) file, 

> text is not suggested on models because of size increasing, anyway in case of text the fonts must be licensed free as the word. 

and about:
> Some of the designs like antenna and holes on the pads had to be taken from Gerber files, by converting them to kicad_pcb and then measuring the dimensions using distance tool.

Gerbv from geda has the ability to measure directly a gerber file, but I think your way does the trick just fine

sandhan26, 2017-11-27:
@easyw Thanks :) . I will push the models, once they are done.

Shackmeister, 2017-12-03:
closing issue as it seems solved :)

closed 2017-12-03 by Shackmeister

sandhan26, 2017-12-08:
[Completed](https://github.com/KiCad/kicad-packages3D/pull/195)