KiCad/kicad-packages3D #175: Resistor_SMD missing

antoniovazquezblanco, 2017-11-29:
Resistors_SMD was added in a8e37f6 and then removed in 39c93cd. What has happened? Was this intended? Should I ask the scripting guys to regenerate them?

Shackmeister, 2017-11-29:
I can see they ended up in Resistors_THT for some reason :/ im gonna generate some new ones in a week or 2 to fix the naming and license

antoniovazquezblanco, 2017-11-29:
Thank you very much @Shackmeister!

grob6000, 2017-12-01:
I kind of quietly renamed this in #180 so I could submit the THT pack, but I guess hang on for the new SMD models...

Shackmeister, 2018-01-04:
PR has been made for the new models:
https://github.com/KiCad/kicad-packages3D/pull/209

pboe, 2018-01-04:
Either the link in the footprints needs to be adjusted or the 3d models need to be renamed.
  (model ${KISYS3DMOD}/Resistor_SMD.3dshapes/R_0603_1608Metric.wrl
  
Filenames 3d model are:  R_0603.step & R_0603.wrl

jkriege2, 2018-01-04:
Models have correct names after merge of #209

poeschlr, 2018-01-04:
@pboe 
Well have you looked into the changes made by @Shackmeister in the comment you answered to? Because that PR will fix the model names.

pboe, 2018-01-04:
Sorry missed that

jkriege2, 2018-01-04:
#209 has been merged!

closed 2018-01-04 by jkriege2