KiCad/kicad-packages3D #186: Geometry check

easyw, 2017-12-04:
Hi,
if FreeCAD (release 0.17 > 12703) can be included in travis checking chain, it would be possible to check geometry of STEP models in an automatic way:
1) modify the following command checking script to give positive or negative results to travis
[checkBOPnoGui.py.zip](https://github.com/KiCad/kicad-packages3D/files/1526396/checkBOPnoGui.py.zip)

2) launch the following command to run the checking in command mode (without the need of a gui)
FreeCADCmd checkBOPnoGui.py "full_path_to_STEP_file/resistor-wrong-BOP.step"

It would be possible then to put some checking at the PR template like:

- [ ] is your 3D model made in a mechanical modeler?
- [ ] have you submitted the source of the model (source file or script)?
- [ ] have you submitted both STEP in scale 1:1 and wrl in scale 1:2.54?
- [ ] are your 3D models aligned to the KiCad footprint?
- [ ] is your WRL model using material properties?

SchrodingersGat, 2017-12-04:
Thanks! I have added a PR template based on your points above. I'll look into the FreeCad script soon

easyw, 2017-12-04:
thx...
I forgot to add an other useful point:
- [ ] have you added a screenshot of the 3D model(s) over the footprint(s) in KiCad?

poeschlr, 2017-12-04:
I think this is a related issue: https://github.com/KiCad/kicad-library/issues/1546#issuecomment-348926505

Shackmeister, 2017-12-04:
Perhaps a Travis check if the model is referenced anywhere?

SchrodingersGat, 2017-12-04:
Do you mean referenced by a footprint?

easyw, 2018-07-11:
No, unfortunately geometry check does not give a clear result. There could be many false results. I wouldn't add any check like this on Travis.