KiCad/kicad-packages3D #207: Error in README

gwideman, 2017-12-27:
... I think:
"Model Scaling [...]    WRL file format [...] KiCad normalizes model parameters to units of Inches,"

It seems to me that KiCad interprets wrl files as 1 wrl unit = 0.1 inch, not 1 inch.

Shackmeister, 2018-01-04:
@SchrodingersGat your phrasing is probably much better than mine but something like:
"Model Scaling [...] WRL file format [...] KiCad normalizes model parameters to units of a tenth of an Inch," ?

SchrodingersGat, 2018-01-06:
@gwideman can you provide some evidence showing that the .wrl files are interpreted with units of 0.1" (100 mil)?

Shackmeister, 2018-01-06:
@SchrodingersGat
http://docs.kicad-pcb.org/stable/en/pcbnew.html#_3d_shapes_libraries_management

From:
13.13. 3-dimensional visualisation

"For small (or very large) packages, a better use of the Wings3D grid is to scale 0.1 inch in Pcbnew = 1 grid unit in Wings3d"

Shackmeister, 2018-01-06:
https://forum.kicad.info/t/wrl-step-import/4043/3?u=shack

SchrodingersGat, 2018-01-06:
Ok, I have fixed this now. Thanks

closed 2018-01-06 by SchrodingersGat

gwideman, 2018-01-06:
@SchrodingersGat  Hi Oliver -- In case you need solid evidence: inspect a wrl file in a text editor for a component whose real-world size you know. For example, Pin_Header_Straight_1x02_Pitch2.54mm.wrl. You will see all coord Coordinate values are in the range of about -1.5 to +3.5 or so. Clearly these are not inches or meters :-).

Better yet, load the wrl file into some software that can read wrl files. I used MeshLab. In MeshLab you can use the tape measure tool to, for example, measure the distance between pin centers. MeshLab shows 1.0 units.