KiCad/kicad-packages3D #211: Diodes_SMD missing

andrewcchen, 2018-01-06:
They are in the [old repo](https://github.com/KiCad/kicad-library/tree/master/modules/packages3d/Diodes_SMD.3dshapes), any reason why they are not here?

jkriege2, 2018-01-06:
They are in Diode_SMD.3dshapes (rename, note the singular!). This repo is meant to work together with https://github.com/KiCad/kicad-footprints which uses the same naming scheme!

Hope that helps,
JAN

closed 2018-01-06 by jkriege2

andrewcchen, 2018-01-06:
Oh sorry that was a typo, I meant LED_SMD. They were in the (LEDs folder)[https://github.com/KiCad/kicad-library/tree/master/modules/packages3d/LEDs.3dshapes] in the old repo.

poeschlr, 2018-01-06:
This repo only accepts models if they have both wrl and step files. The old wings stuff can no longer be used.

SchrodingersGat, 2018-01-06:
Yes these will need to be re-generated from scripts

antoniovazquezblanco, 2018-01-25:
Sorry to be bothering you here.
LED_SMD footprints are still missing in this repo.
Could we keep this open until the issue is fixed?
I prefer to find the issue open rather than having to look for it and double checking if it supposed to be closed.

poeschlr, 2018-01-25:
I don't think we should have an issue open for each missing 3d model (not even for each missing set of models.) Unless they did exist in the old repo as step files.

antoniovazquezblanco, 2018-01-25:
Sorry about that. You are right. Didn't realize step models where not available in the old repo.

Thank you again @poeschlr.

samyhocine, 2018-03-29:
Hi,

Sorry poeschlr but I think this is an issue. KiCAD 5 is on it's way and Led_SMD is still missing. In the sources, there is only a freecad project file for the 1206 version.

I just want to be sure the community will have a Led_SMD folder when KiCAD 5 will be out.

Of course it's not a critical issue but it's still an issue.

Thanks ;)