KiCad/kicad-packages3D #219: the file LED_D5.0mm.step behaves badly

joojaa, 2018-01-10:
The file **Resistor_THT.3dshapes\LED_D5.0mm.step** behaves badly when exported to a step file form pcbnew. I have tested the generated file in following mCADs: Siemens NX, PTC Creo and Solidworks. I am still unable to pinpoint the problem actual as the original file works fine at least when tested in Creo. Replacing the file with a new STEP file, however fixes the problem. 

I havent had time to debug the STEP file contents but seems like there is some problem with either the top dome or the cylindrical side. Now it is likely that the problem is in the exporter of Pcbnew. The exporter has other problems also as it is a bit sub optimal generating more sub assemblies than needed. However, changing the file eliminates this problem so might be worth doing?

[image]

**image 1**: Attached screenshot showing creo and NX side by side

jkriege2, 2018-01-10:
What changes need to be done to the file?

Shackmeister, 2018-01-10:
Has the file been fused properly before export?

poeschlr, 2018-01-10:
I checked the file using the tools available in freecad and the step file on its own seems to be correct. (It is fused correctly and the geometry test does not complain about anything.)

How did you export the step model of the board? (kicad or freecad+stepup? which version?)
Are you sure you use the current step model found in this repo?

joojaa, 2018-01-10:
Exported using Kicad version 2018-01-10 revision dc62c0717

Like I said the individual file is fine its just broken once exported by kicad. My other random sampling did not have this problem.

>> Are you sure you use the current step model found in this repo?

Yes, loaded the file 15 minutes before test.

>> I checked the file using the tools available in freecad and the step file on its own seems to be correct.

That may be but freecad is not, as far as I am aware part of the STEP conformance testing group so it does not really count.

poeschlr, 2018-01-10:
If the individual file is ok but the export breaks it, then this is the wrong place to report this. That would indicate a bug in the step exporter.
Such bugs should be reported over at the launchpad bug tracker: https://bugs.launchpad.net/kicad
Include your testfile and your version information in the report.

joojaa, 2018-01-10:
Yes but ist not entirely clear wether the bug is here or there, it may be that the bug is in fact in both. Or just one, I can not rule this out until I have had time to manually parse the step file.

poeschlr, 2018-01-10:
Well you could try to import the board into freecad using stepup and export it as a step file from there. If it works with that workflown, then the probability is quite high that the problem is indeed in kicads internal step exporter.

easyw, 2018-01-11:
> I checked the file using the tools available in freecad and the step file on its own seems to be correct. (It is fused correctly and the geometry test does not complain about anything.)

the 'LED_D5.0mm.step' has some issue tested with FC0.17
[image]
[image]
@jkriege2 changing the formula for the first Pad to 'Spreadsheet.din / 1.98' will remove the issue
[LED_D5.0mm-step.zip](https://github.com/KiCad/kicad-packages3D/files/1621914/LED_D5.0mm-step.zip)
@joojaa 
> The exporter has other problems also as it is a bit sub optimal generating more sub assemblies than needed.

Still it would be nice to test the exporter with StepUp ... it shouldn't have any extra assemblies

Shackmeister, 2018-05-02:
is this issue still present?

joojaa, 2018-05-03:
@easyw yeah that is indeed better

antoniovazquezblanco, 2018-10-02:
@joojaa Can we close this issue then?

joojala, 2018-10-02:
@antoniovazquezblanco  sure