KiCad/kicad-packages3D #221: Could you help me identify the name and source of this PushButton SMD

fandres, 2018-01-15:
Hi, I wanted to contribute to this fabulous project. I have modeled this, switch: button SMD switch 3x6x2.5mm, this modeled in FreeCAD, in one piece, they are already in step format and wrl with vrld materials, I know I have to upload the source file in
kicad-packages3D-source. The problem is that apparently the switch is generic and I can not trace the source of origin, the switch is very evolved in web pages of electronica and very used. **Could you help me identify the name and the source of it?** Thank you

https://www.ebay.com/itm/15Pcs-3x6x2-5MM-Momentary-Tact-SMD-SMT-Push-Button-Micro-Switch-2-Pin/322918285966?epid=1462083428&hash=item4b2f6e028e:g:siUAAOSwyYFaIwi3
https://es.aliexpress.com/item/Free-Shipping-1000pcs-lot-Tactile-Push-Button-Switch-3x6x2-5mm-SMD-Tact-Switch-2-pins-micro/32216027178.html
[image]

diggit, 2018-01-15:
[This one seems similar](https://cz.mouser.com/productdetail/ck-components/rs-282g05a3-sm-rt?qs=sGAEpiMZZMsgGjVA3toVBERYOiiKsymw96bNzPZcDJQ%3D)...

fandres, 2018-01-16:
In effect, I will leave that, there are others very similar.
SW_SPST_RS-282G05A3-SM RT
The name is correct? Is there any special notation for this?