KiCad/kicad-packages3D #229: Do we want to include the EP size in the 3d model names?

poeschlr, 2018-01-22:
Do we want to have multiple 3d models for different ep sizes or do we map all differing footprints to only one model?
(Currently most footprints only exist in one ep size. But i suspect that now that we have the size included in the name, a lot of contributions for more specialized footprints might be made.)

easyw, 2018-01-22:
For BGA we have decided to include only the balls on the perimeter and not inside because of the file size...
https://forum.kicad.info/t/3d-new-library-for-mechanical-cad-exporting-and-enclosure-design/1763/276
the user can generate the full balls model from the scripts on request.
Many users are already questioning on a 3D library file size, so I wouldn't add all the sub-models to the standard library... The user can always generate an accurate model from the scripts, in case of need. Moreover 3D model is just a reference to check the footprint. Best is by designing it accordingly to ds specification.
KiCad Library man power is very limited ATM...

poeschlr, 2018-01-23:
Ok in that case i will add the ep size to the models we already have. (use the rename information i will also use for the symbol modification.)
Otherwise these models would not really make sense and we would need to create models without exposed pads. (A lot of footprints currently only exist as ep versions.)

closed 2018-01-23 by poeschlr