KiCad/kicad-packages3D #230: Why are models unioned?

sethhillbrand, 2018-01-24:
Probably a simple question but I couldn't find the answer in the usual places:
http://kicad-pcb.org/libraries/contribute/
https://github.com/KiCad/kicad-packages3D/wiki/Contributing

Why do we want to have the models unioned before exporting to STEP?

-Thanks!

easyw, 2018-01-24:
The initial concept was to have one single part each model. 
Most of the manufacturers are also using this convention. Simplicity is the best.
The other main reason is that FreeCAD 0.15 and FC 0.16 don't have support for an assembly container.
Then i.e a QFP model would be made by hundreds of pins, a body and a pin mark instead of a single handleable object. Moreover the board assembly would have been a bunch of thousand pieces, instead of a collections of board and components.
Because FreeCAD is the sw used to build the libraries manually or through scripts,  having the model unioned has been the right choice since the beginning of this KiCad MCAD library project.
You can have your answers at the KiCad forum
https://forum.kicad.info/search?q=step
https://forum.kicad.info/t/3d-new-library-for-mechanical-cad-exporting-and-enclosure-design/1763
or at the developers mailing list.

sethhillbrand, 2018-01-26:
Thanks for the info!  Good to know the history.

closed 2018-01-26 by sethhillbrand