KiCad/kicad-packages3D #247: Radial capacitors have no height

strombom, 2018-01-29:
Is there a reason that the radial capacitors have diameter and pitch, but no height? All seems to have the same length as diameter.

Example:
CP_Radial_D10.0mm_P3.50mm

The axial capacitors have diameter, pitch and length, example:
CP_Axial_L93.0mm_D29.0mm_P100.00mm_Horizontal

Is this by design or a mistake?

evanshultz, 2018-01-29:
This is a good question and something that has come up in discussions at the old repos.

Quick answer: Radial caps do not require height to define the footprint, while axial caps do.

Adding the various heights for each radial cap size (diameter and lead pitch) would result in dramatically more footprints and 3D models. However, without having height specified by the selected footprint the actual capacitor height in the 3D viewer is not connected with the real part selected by the user. (Another consideration is that for a given electrical spec there may have a range of heights available.)

Personally I'd like to include height, at least just for the visual representation and manual checking. But since KiCad's design constraint system is so weak there's not much that can be done by having the actual height to automatically capture any part collisions, chassis collisions, or other mechanical checks. So it seems a bit hard to make this a high priority in my opinion.

Shackmeister, 2018-01-29:
Im currently redoing the radial smd caps footprints and models.. I can add heights if @poeschlr and the other librarians would agree on it

strombom, 2018-01-29:
I started going through Jamicons datasheets to make a table of all DxLxp combinations. Then maybe Nichicon and some more. Would that be of any use or is this information already available somewhere?

strombom, 2018-01-30:
Here is a list of capacitor sizes. The height is max height including tolerance, that's why there are many variants, maybe they should be rounded up to closest 1mm or 0,5mm.

https://pastebin.com/SqtPhgfM

jkriege2, 2018-01-30:
I scripted the footprints (and the first, now obsolete round of semi-scripted 3D models), I decided against the height, as I wanted to arrive at a useful set of footprints. That was also before the proper mechanical CAD library, so 3D models were more for nice representation, not so much for exactness. As @evanshultz said above, therefore I skipped the height (actually anything that was not necessary to specify the footprint) and took a "representative" height (whatever that is) for the 3D models.

I think we have several comparable cases: 
- There should be vertical power-resistors that might have the same problem
- For LEDs there are some designed to be stacked at the PCB-edge, there a height (or Z-elevation) _Z and a pin-LED-offset _O is given (see https://github.com/KiCad/kicad-footprints/blob/master/LED_THT.pretty/LED_D3.0mm_Horizontal_O1.27mm_Z6.0mm.kicad_mod) ... but then we should also duplicate the LEDs for colors (I started recently when reworking OptoDevice to have specific clear/IRGrey/IRBlack-LEDs for photo-diodes and IR-LEDs, see https://github.com/KiCad/kicad-footprints/pull/308)

jkriege2, 2018-01-30:
We could start out by adding the height to the existing footprints/models and then go on by extending the range of available footprints. But bear in mind that this might really explode the lib of footprints!

antoniovazquezblanco, 2018-07-25:
I don't want to get offtopic but it would be desirable to have a 3D model selection tool in the same way footprints are chosen.

What do you think about that feature? It would avoid footprint duplicity. It would also fix the LED color problem in models.

Shackmeister, 2018-07-25:
We discussed it somewhere I cant remember and we (@evanshultz) made a bug on launchpad.

antoniovazquezblanco, 2018-07-25:
Bug report by @evanshultz:
https://bugs.launchpad.net/kicad/+bug/1766323

Another related bug?
https://bugs.launchpad.net/kicad/+bug/1685004

Given that the feature was already requested I don't see the point on generating a ton o footprints to solve this issue to later delete them. Can we plan this in advance taking into account that the feature may be available in the future?

poeschlr, 2018-07-25:
The earliest this can be added is with v6 which by my estimate is 2 years away. Maybe more depending on how long it takes to get 5.1 out. So right now we should work as if this feature will never arrive.

antoniovazquezblanco, 2019-05-08:
I am working on an extremely tight design that needs case manufacturing. I need to use the appropriate 3D models for this and I could not find the desired 3D model in the repo.

I was wondering if a PR that would rename THT cap 3D models to include the actual height would be accepted. @poeschlr ?

The idea would be to allow me to later submit new models that the user may choose from in case they need it.

antoniovazquezblanco, 2019-07-18:
Related issue https://github.com/KiCad/kicad-footprints/pull/1715