KiCad/kicad-packages3D #248: Nominal or max size

strombom, 2018-01-30:
Just a question regarding 3D models:

Should we use nominal or maximum dimensions?

For example, if height of a component is specified to 10+-1 mm, should I make the component 10 mm or 11 mm high in the model?

easyw, 2018-01-30:
I think everything has been developed with nominal dim's.
The tolerance has to be applied also to enclosure, so a collisions checker as well measurements of gaps are very useful in mechanical design.

strombom, 2018-01-30:
Sounds reasonable, I also checked some manufacturer step files and they use nominal sizes too. Thank you.

Shackmeister, 2018-05-02:
should this be added in the readme?

antoniovazquezblanco, 2018-07-12:
I would add this to the readme just in case anyone has this doubt again.

evanshultz, 2018-07-12:
@antoniovazquezblanco Good point. Added at https://github.com/KiCad/kicad-packages3D/pull/340.

antoniovazquezblanco, 2018-07-13:
Thanks!

Could this be closed?

closed 2018-09-10 by Shackmeister