KiCad/kicad-packages3D #254: 0805 led rotation error?

pointhi, 2018-02-02:
Just relying this from irc:

https://i.imgur.com/IjHrLKj.png

I have no system to recheck this at the moment. If it's a problem with old and new library use at the same time please simply close.

evanshultz, 2018-02-03:
@pointhi 
Just to clarify, it looks like the green dots on the 3D model are opposite the silk mark for pin 1. FYI, all 0805 LED footprints have the correct 3D model parameters (scaling = 1, location = 0 rotation = 0).

evanshultz, 2018-02-03:
And here's the image for easier reference:
[image]

poeschlr, 2018-02-03:
This repo has no 3d models for smd leds. I think @Shackmeister started to make some but i think they have never been merged.

Shackmeister, 2018-02-03:
I have just picked up the project again. I finished porting the stript to yaml. So models should be here next week

poeschlr, 2018-02-03:
@pointhi 
We do not intent to transfer the old wings models into this step based lib.
For this reason i think we can close this as it clearly does not apply to the new 3d model lib.

closed 2018-02-03 by poeschlr