KiCad/kicad-packages3D #255: Capacitor_THT C_Axial_L12.0mm_D6.5mm_P20.00mm_Horizontal have different X offset for wrl and step models

eldarkg, 2018-02-03:
See Capacitor_THT.3dshapes/C_Axial_L12.0mm_D6.5mm_P20.00mm_Horizontal.step and .wrl files
[image]
Therefore export of pcb board to STEP is wrong.

poeschlr, 2018-02-03:
Where did you get the footprint that is defined that way?
The current footprint only has the wrl file added as it should be.

eldarkg, 2018-02-03:
@poeschlr I `Add 3D Shape` STEP manually for the issue visibility

eldarkg, 2018-02-03:
@poeschlr WRL and STEP models have a different origin point

poeschlr, 2018-02-03:
It seems all step models for capcitors have the wrong origin.

The strange thing is the current script generates them correctly.

easyw, 2018-02-03:
@poeschlr 
that was because the models had been generated by a bugged FC release.

poeschlr, 2018-02-03:
Yea i thought so. I am currently working on integrating the geometry check into these generators and will export new models shortly.

easyw, 2018-02-03:
@poeschlr 
please consider that, when there are fillets, FC 0.17 BOP checker may give false positives...
Here a long thread about it:
https://forum.freecadweb.org/viewtopic.php?f=8&t=25894

In case of check failure I use to double check the models manually with Design Spark Mechanical and then open the models also in CAD Assistant and Varicad to see if there are some visual issues...

poeschlr, 2018-02-03:
Fix for these is committed. https://github.com/KiCad/kicad-packages3D/pull/256
I fear others might be wrong as well but that has to wait for another day.

eldarkg, 2018-02-04:
@poeschlr I think it is need to check another models in the repo for this issue too.

commit bdc2e1a5 (2018-04-24): Merge pull request #298 from poeschlr/fix_step_alignment_cp_axial

Fix #255 for CP_Axial models. (Step models where alignd wrong.)
commit d57c28b9 (2018-04-24): Fix #255 for CP_Axial models. (Step models where alignd wrong.)
closed 2018-04-24 by evanshultz