KiCad/kicad-packages3D #259: Connector_USB: 3D models are missed

eldarkg, 2018-02-10:
The old package contained 3D models of USB_A, USB_B and etc

poeschlr, 2018-02-10:
Are these 3d models available as step files? If not then we can not move them over here as we require this lib to be fully step compatible.
In addition we also require the wrl files to be scaled correctly which is unlikely if there is no step file for them.

evanshultz, 2018-02-10:
In addition, those USB connectors should be removed (see https://github.com/KiCad/kicad-footprints/issues/290). Unless you can find the manufacturer and MPN so they're not generic.

Shackmeister, 2018-05-02:
closing since all mechanical models avaliable are in the repo

closed 2018-05-02 by Shackmeister

eldarkg, 2018-05-02:
@Shackmeister It is available only micro usb models