KiCad/kicad-packages3D #275: Scripts broken

esden, 2018-02-25:
The README refers to  https://github.com/easyw/kicad-3d-models-in-freecad as the model for how to add new 3d models to Kicad library. The scripts in that repository are all broken and do not work out of the box. The Documentation is misleading and the scripts it refers to don't even exist.

Is that repository actually the best resource for adding new 3D Models to KiCad default libraries?

poeschlr, 2018-02-25:
What do you mean these scripts are broken? They worked fine for creating the models in this repo ;)

esden, 2018-02-25:
Maybe in the very specific setup of the one person that ran them to generate the current models. The scripts definitely don't work on a default Ubuntu xenial install with current freecad 0.15 install and most recent cadquery plugin.

Have you tried it recently?

This makes it impossible for anyone besides the one person with the working setup to run the scripts to add new models.

You can refer to my other issues I have raised on the repository to see what issues I am having: 
 * easyw/kicad-3d-models-in-freecad/issues/156
 * easyw/kicad-3d-models-in-freecad/issues/157
 * easyw/kicad-3d-models-in-freecad/issues/158

easyw, 2018-02-25:
> This makes it impossible for anyone besides the one person with the working setup to run the scripts to add new models.

so how you can match your statement with the other contributors of the repo?

> The scripts in that repository are all broken and do not work out of the box. 

Instead of simply complain about all, you could ask for help in having a working environment with a simple 3d family, posting your errors/doubts
The scripts depend on FC and CQ versions... the chain may be or not broken due to recent release of CQ, not tested recently, but anyway IMO, your attitude is not the right one to contribute anyway.
 
Cheers
Maurice

Shackmeister, 2018-05-02:
closing this since there havent been a follow up.. and the scripts seems to be working fine for most other contributors

closed 2018-05-02 by Shackmeister