KiCad/kicad-packages3D #278: The compiled Kicad-pacakges3D library package should be split into multiple part files

sandhan26, 2018-03-08:
In many Organizations, internet is accessed via a proxy server, which have a download limit on file size. 
The Kicad-pacakges3D library package is very large in size, an so it fails to download through a proxy internet connection.

I had to download the package manually and install it.

To avoid the above problem, the Kicad-pacakges3D library package can be split into multiple files of 100MB and put into the repository.

Kindly advise.

easyw, 2018-03-08:
https://kicad.github.io/packages3d/

sandhan26, 2018-03-08:
I use ppa:js-reynaud/ppa-kicad on my Ubuntu 16.04. So, as while upgrading the packages with `sudo apt get upgrade -y` the 3d packages get rejected by the proxy server.

@easyw @poeschlr I could download the package manually, but it would have been much simpler if one could simply upgrade their packages with the aptitude package manager.

Ratfink, 2018-03-08:
Oh, I understand now that this issue is about distribution packaging.  We're not in control of how distros' package managers work, and this isn't the right tracker to report issues with distro packaging.  If you want the packages available with `apt-get` to be split into multiple smaller packages, you should bring that issue up with [js-reynaud](https://launchpad.net/~js-reynaud) directly.

closed 2018-03-08 by Ratfink

poeschlr, 2018-03-08:
The packages for v5 are already split up. If a user has problems with file sizes they can choose not to install the 3d models package but instead download (manually) only the models they need.

sandhan26, 2018-03-08:
Thanks guys :+1: