KiCad/kicad-packages3D #286: LED_SMD still missing

samyhocine, 2018-03-29:
Hi,

Sorry poeschlr but I think this is an issue. KiCAD 5 is on it's way and Led_SMD is still missing. In the sources, there is only a freecad project file for the 1206 version.

I just want to be sure the community will have a Led_SMD folder when KiCAD 5 will be out.

Of course it's not a critical issue but it's still an issue.

Thanks ;)

easyw, 2018-03-31:
@samyhocine 
we have already a 3d model generator for SMD LEDs
https://github.com/easyw/kicad-3d-models-in-freecad/pull/155
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/LED_SMD
If you are willing to check, we could have those merged to this repo 😄

Shackmeister, 2018-05-02:
LED SMD 3D models has been merged here https://github.com/KiCad/kicad-packages3D/pull/280

closed 2018-05-02 by Shackmeister