KiCad/kicad-packages3D #296: Wish: Add component source reference in documentation

KammutierSpule, 2018-04-16:
Would be possible to add the references
eg: documents from original manufactors
where the models were based?

It could be a link URL to some examples of products that the models will apply.

This way I can compare and make sure I am selecting the right model for the component I will be using.

KammutierSpule, 2018-04-16:
Hi Jan,
I knew about 1. and 3. but didn't check the 2.
Thanks for the suggestion I will have a look on that.

I understand that it may makes more sense for this library from the use of footprint perspective (as footprints has 3D models assigned)
but this need was come to me on the other way as I was making a composite of 3D models for some custom design and I was look in the existent 3D models from this library.

antoniovazquezblanco, 2018-07-25:
@KammutierSpule Have your needs been addressed or do you have a suggestion on how to link a part with the documentation in a better way?

Thanks

KammutierSpule, 2018-07-25:
I didn't checked it again recently. I believe it would be useful to add references as much as possible on every elements ( symbol, footprints, 3D shapes, etc). I will leave it close by now.

antoniovazquezblanco, 2018-07-25:
The way I see this is that every 3D file format has its own way of adding comments and it is not efficient to add the source in each file in a different way.

Given this the only way to proceed I see is to create another file which holds the information about the models and even references to the models themselves, sort of what `CREDITS.md` does but if possible it would be nice to use a format that can be used programmatically.

Of course this would need support from the devs and librarians. I agree that it would be nice to standardize the way in which information is held in this repo even if this is not yet supported by KiCad itself.

Is that what you are looking for?

KammutierSpule, 2018-07-25:
Yes, also, that will make possible if a user wants to confirm / to validate the things. Or if the user does not understand (i.e: totally agree with) the element, it can be checked against the reference.

KammutierSpule, 2018-07-25:
oks nice! lets keep it open for discussion.