KiCad/kicad-packages3D #299: Missing potentiometer models

Ratfink, 2018-04-24:
As Wayne reported [here](https://lists.launchpad.net/kicad-lib-committers/msg00547.html), the [potentiometer 3D models from 4.0](https://github.com/KiCad/kicad-library/tree/master/modules/packages3d/Potentiometers.3dshapes) are all missing in the new library.  I know that all the old models made with Wings had to be discarded, but many of those old potentiometer models were made with FreeCAD, so could probably still be salvaged.

Misca1234, 2018-09-23:
I looked into some of them

I guess few of them can be salvaged without more work

They are almost always positioned wrong and they missing parts, some of them are difficult to re map back to new file names

Potentiometer_Alps_RK09L_Single_Vertical, positioned wrong, missing metal "holder"
Stepup macro, no modifications
[image]

Potentiometer_Omeg_PC16PU_Vertical, positioned wrong
Stepup macro, no modifications
[image]

Misca1234, 2019-02-22:
@Ratfink 
Shall we close this thread ?

Misca1234, 2019-06-30:
@Ratfink 
Do you think it is worth exploring the possibility to salvage the old ones?