KiCad/kicad-packages3D #315: Missing Hirose 3D models

jorticus, 2018-05-18:
The Hirose parts in Connectors_FPC-FFC.pretty reference models which seem to be missing from this repository.

Eg.
https://github.com/KiCad/kicad-footprints/blob/master/Connector_FFC-FPC.pretty/Hirose_FH12-10S-0.5SH_1x10-1MP_P0.50mm_Horizontal.kicad_mod
Specifies
`${KISYS3DMOD}/Connector_FFC-FPC.3dshapes/Hirose_FH12-10S-0.5SH_1x10-1MP_P0.50mm_Horizontal.wrl`
which is nowhere to be found!

evanshultz, 2018-05-18:
We always add the 3D model reference to all footprints, even if the 3D model doesn't exist, so if/when the 3D model is added the footprint doesn't need to be touched.

For these particular 3D models, I suspect they simply haven't been created.

closed 2018-05-18 by evanshultz

jorticus, 2018-05-19:
Ah thanks, that makes sense, though it's a little confusing since Kicad throws an error everytime you open the 3D PCB viewer.