KiCad/kicad-packages3D #330: Ublox 3D models

Shackmeister, 2018-07-03:
questions for anybody with a knowledge about license (@easyw @poeschlr @SchrodingersGat )
Perhaps the Ublox commiters can answer this themselves (@mazgch @stefanobianconi @uBloXAE @fredrikbrodjeublox)

I was looking at the Ublox models provided at https://github.com/u-blox/3D-Step-Models-Library
The readme says the following:
Copyright (C) u-blox

u-blox reserves all rights in this deliverable (documentation, software, etc., hereafter “Deliverable”).

u-blox grants you the right to use, copy, modify and distribute the Deliverable provided hereunder for any purpose without fee.

THIS DELIVERABLE IS BEING PROVIDED "AS IS", WITHOUT ANY EXPRESS OR IMPLIED WARRANTY. IN PARTICULAR, NEITHER THE AUTHOR NOR U-BLOX MAKES ANY REPRESENTATION OR WARRANTY OF ANY KIND CONCERNING THE MERCHANTABILITY OF THIS DELIVERABLE OR ITS FITNESS FOR ANY PARTICULAR PURPOSE.

In case you provide us a feedback or make a contribution in the form of a further development of the Deliverable (“Contribution”), u-blox will have the same rights as granted to you, namely to use, copy, modify and distribute the Contribution provided to us for any purpose without fee.

Is this enough for us (Kicad) to be allowed to relicense and redistribute these models in the Packages3D repository?

antoniovazquezblanco, 2018-07-03:
I think we would need the explicit permission to relicense the models under CC-BY-SA 4.0 wich is the only license accepted ATM for 3D model contributions.

Am I right?

Misca1234, 2018-09-23:
> I think we would need the explicit permission to relicense the models under CC-BY-SA 4.0 
> wich is the only license accepted ATM for 3D model contributions.
> 
> Am I right?

I want to know this as well because I am involved in a discussion with a contributer that want to take the step file from manufactures

poeschlr, 2018-09-23:
Any model in the official lib must be relicensed under our license. It does not make sense for us to open up here as it will only increase user confusion. I am aware that some users might find it useful to have an already assembled source of these manufacturer models. 

It is free to anybody to open a third party repo where models with strange licenses are collected. (but this repo can not be under direct control of kicad. We can also not distribute such models with kicad.)

antoniovazquezblanco, 2018-09-25:
I think the way to move fordward is the following:

1. Either the manufacturers publish their models under the same license explicitelly in their corresponding webpages.
2. Either the manufacturers send their files to the repo agreeing to the relicense with CLA bot.

What do you think? Is the question solved?

Misca1234, 2018-09-25:
I think it would be difficult to get them to send them in, expecially if we ask them to remake (rescaling and use right colours)
well, unless we will do the work on them afterwards.

Least problems would be just to say no to anything that originates form the manufacture.

closed 2019-01-19 by Shackmeister