KiCad/kicad-packages3D #358: Request for full size USB A and B connectors

BrendanSimon, 2018-08-04:
Request for full size USB A and B connectors please :)
I could only find Micro and Mini B connectors.

closed 2018-09-10 by Shackmeister

deltabeard, 2020-02-24:
Why was this closed? There are no 3D models for USB A and B connectors available.

evanshultz, 2020-02-24:
I can't speak for @Shackmeister , but as a rule we don't take requests. All librarians are volunteers and the library is largely built from community contributions. There are lots of missing elements and so if anyone sees something missing that they want we welcome them to contribute it. In a form that matches our license, of course, which can be an issue with 3D models.

I would like to see the librarian leave a message when closing an issue, however. Perhaps this was just a mistaken click?