KiCad/kicad-packages3D #366: SO-4_4.4x4.3mm_P2.54mm.wrl missing

romain145, 2018-08-14:
The footprint Package_SO:SO-4_4.4x4.3mm_P2.54mm refers to the 3D model SO-4_4.4x4.3mm_P2.54mm.wrl which is not shipped with KiCAD 5.0.0 and doesn't appear to be here either.

Can you point me where to find this 3D model?

Misca1234, 2018-09-23:
It is not mandatory to supply a 3D model with a foot print 
but it is mandatory for a foot print to "point" to its corresponding 3D model.

Are yo interested in how to make 3D models using a parameter file to generate 3D models?
There are python scripts that read a parameter file and generate the 3D model.
See following link for an example.

3D model push itself (these files are generated by the script)
https://github.com/KiCad/kicad-packages3D/pull/393

Parameter push (this is what you add)
https://github.com/easyw/kicad-3d-models-in-freecad/pull/191

Misca1234, 2018-10-02:
The 3D model is ready for merge

closed 2018-10-02 by Misca1234