KiCad/kicad-packages3D #369: can we have small size version?

walkthetalk, 2018-08-16:
on archlinux:

Package Size: 289.2 MB
Install Size: 4.3 GB

so many space, can we have other file types, for zipped version to be installed?

evanshultz, 2018-08-16:
你好！不可以，对不起.。
https://lists.launchpad.net/kicad-developers/msg36982.html
https://github.com/KiCad/kicad-winbuilder/issues/53

Not now, but hopefully in the future.

poeschlr, 2018-08-16:
I assume that the 3d models are a separate package on arch. I doubt that they can be compressed to such a low value so i guess the ~200MB is the main kicad package not the package for the 3d models.
So if you do not want the 3d models you could simply not install that package. You can then download the 3d models that you need manually from the download page. (We offer single libs packaged as zip files.)

Misca1234, 2018-12-08:
I guess the thread can be closed?

eujean, 2018-12-14:
Perhaps this is a problem?

Total Download Size:    291.53 MiB
Total Installed Size:  4449.22 MiB

Note this is only for the 3d-packages, not including the kicad installer itself which is approximately 200 MiB.

I'm just curious how the models are stored, and how they are compresses at a compression rate of almost 95%.

poeschlr, 2018-12-14:
Step files have quite some redundancy in them. If you then not only compress single files but full directories then you get quite a lot of potential. (After all: Step is human readable. A binary format would be much much smaller but that makes portability a lot harder to achieve. There is a reason step is the king in the mcad world.)

However this is only practical if you only need to compress and decompress them once. As always there is simply a tradeoff between computing power and storage. In todays world where storage is basically free one can optimize for computing speed. (Or would you rather wait 5 minutes everytime you open the 3d viewer till the 3d models are decompressed?)

If storage is a real concern than simply do not install the 3d models via the installer but only download the ones you really need.

Franck78, 2020-09-08:
Why do I loose my time reading issues that are not issues ????

Someone to close that please ?

chschlue, 2020-09-08:
This _is_ an unresolved issue. If you don't like it, don't read it.

Also, you've been asked to work on the general tone of your comments before. Heed that advice.

Franck78, 2020-09-08:
I'm sorry but you have to read "a lot" of issues or thing that are not real issues but "things" sitting there.

example:
footprint merged:
https://github.com/easyw/kicad-3d-models-in-freecad/pull/351

corresponding model waiting who knows what
https://github.com/KiCad/kicad-packages3D/pull/683

hildogjr, 2020-09-30:
I would like to invite @sethhillbrand to this discussion. He made a nice [study on reduce the STEP file size](https://fosdem.org/2020/schedule/event/step_reduce/), having a [algorithm to do so](https://gitlab.com/sethhillbrand/stepreduce).

As far I tested in some STEP 3D models of KiCad, I got a good file size decrease (actual 3D files appear that doesn't passed to this process). Also I would like to mention the [compressed step GZ format capability](https://gitlab.com/kicad/code/kicad/-/issues/2479) and [compressed step ZIP](https://gitlab.com/kicad/code/kicad/-/issues/5376). Could we clean the STEP for v6 (@sethhillbrand algorithm) and maybe compressed for v7?

sethhillbrand, 2020-09-30:
I've been down this road before (https://github.com/KiCad/kicad-packages3D/pull/223)  That was an enormously frustrating conversation that led nowhere productive.

If this is desired, I'm happy to provide pipeline code in Python to compress our models and verify the compression results (comparing model results in OCC).  But I'm going to stay out of the discussion about whether this should be done.  I believe it should but I don't want another disagreement with anonymous internet experts.

Franck78, 2020-10-01:
If there is a very little thing to do to the .step files, it's a check that 'filename' and 2 others fields are correctly filled. 

Lot of PR are blocked or retarded because 'blah blah filename in step not good'.

The reason for this field is 'ease matching something with something else in some CAD'. Clearly outside of KiCAD.
A monthly check can fix all the new 3d step added. No one will cry if the 'match' is wrong during a month for a new 3dimage

That will save *a lot of time*  here.

chschlue, 2020-10-01:
@hildogjr @sethhillbrand If anything, I would delay stepreducing the files, not the other way round.

Deflating the models should be safe in any case and is also reversible.
I also don't know why stepreducing would be a problem.

hildogjr, 2020-10-01:
I made some test with third-part model from https://grabcad.com/ and https://www.3dcontentcentral.com/. I also used the `stepreducing` algorithm to reduce some National Instruments 3D models of 100MB, finishing with a 83MB. I had no problem with Pcbnew or FreeCAD in an of those cases.

Now, using Nightly KiCad, I am also compressing those models into `.STEP.GZ` (using the extension .stpz, cause it's the default for FreeCad). Again, no error or warning on both softwares, even using the exported board STEP model, `stepreduce`-ing it and compressing into GZ, and import into FreeCAD.

I just saw gain in the loading time of FreeCad.

Franck78, 2020-10-01:
```
ATZ
ATL0M3B0S7=30F1%C3
ATDT18003317166
```
If you are older than 50, you may remember that.

What is this ? Simply order the v42bis MNP5 modem to call the Compuserve service. With compression enabled (%C3)

Funny to see that after 40 years, problem is not solved despite compressed filesystems available (zfs, ntfs,...)