KiCad/kicad-packages3D #379: License for 3d model on manufacturer website

penoud, 2018-08-23:
I want to know if the Kicad team approach manufacturers to discuss about the opportunity to setup a agreement on the use of 3d  model for Kicad? 

This idea came to me for a Molex connector. The STEP is provided on their website. But the legal term did not allow to use this model for Kicad. The design is quite complex and this will help a faster grow for the libraries.

pointhi, 2018-08-24:
This only works when you manage to convince them to publish their models in a KiCad compatible license.

I did some talk with [Würth](https://www.we-online.de) some time ago, and they agreed with redistribution. But there were some constraints attached which in fact made them incompatible with the KiCad license. I suspect that there are many companies which would relicense their models.

Rather it would be way easier to have a separated package3D repositories with different licensing terms where commercial models live in. This could be something companies could agree to. At least KiCad seems to supports multiple 3D-Model search paths.

penoud, 2018-08-31:
Dear pointhi. Thank you for your elaborate answer. This help me to understand how this process could be complex. I was not realising that the manufacturer should not only open a "little bit" his license but he should comply with Kicad one. 
Then I would be a long discussion process to conveince them to switch their license. Then I would be easier to redesign everything from scratch. That sade.

closed 2018-09-02 by penoud