KiCad/kicad-packages3D #405: Metres are the unit of measurement in VRML

bartmcleod, 2018-09-20:
Your documentation states that WRL does not specify units of measurement and that you have adopted inches as the default. As far as I know, metres are the standard for VRML and there is indeed no way to specify anything else.

Misca1234, 2018-09-23:
If I do not recall wrong the .wrl file is dimensionless but the overall strong recommendation on the planet is metric.

> have adopted inches as the default.

Yes, it is a bummer everything gets messed up
When I digged into this for a year ago or so, it seems it had historic reasons from the time when FreeCad was not widely used but instead Wings 3D to make the 3D models and the internal format of the 3D model files was read in as they was.

bartmcleod, 2018-09-23:
Thank you for the clarification.

closed 2018-09-23 by bartmcleod