KiCad/kicad-packages3D #408: ESP32-WROOM-32 issues

Cylindric, 2018-09-25:
Hi folks. I was just about to create an Espressif ESP32-WROOM-32 part, and found this 3D package already exists. I have a couple of questions about it though:

1. there seems to be no symbol using it
1. there seems to be no footprint using it
1. Is there a reason that it shows as a blue board with orange terminals for me? It looks like the materials are a bit messed up. It should be a black PCB with gold terminals.
1. It doesn't seem to be centred in such a way that an KLC-compliant footprint doesn't complain about F6.2. It needs nudging down 50mils.

I'm happy to submit the symbol and footprint for it, as I have created it for my own use already - they just need some KLC-love.

[image]

[image]

Cylindric, 2018-09-26:
Aah, somehow missed that footprint. Was looking in Modules and not RF_Modules. Be nice if the modules matched the other categories, and it was `Modules_RF`.

Anyway, I guess the other questions still apply.

diegoherranz, 2018-10-16:
> there seems to be no symbol using it

It's pending merge here https://github.com/KiCad/kicad-symbols/pull/724

> Is there a reason that it shows as a blue board with orange terminals for me? It looks like the materials are a bit messed up. It should be a black PCB with gold terminals.

Yes, I guess that could be improved

Misca1234, 2018-10-27:
The blue colour comes form that the board was blue before, 
https://www.bing.com/th?id=OIP.Ucyp18rOnTjbudf2pUW2-wHaFj&w=288&h=205&c=7&o=5&pid=1.7

now they are white
https://www.espressif.com/en/products/hardware/esp-wroom-32/overview

The reason for "orange" colour on the terminals it is that how copper looks like when you select it in the kicad stepup macro

easyw, 2018-10-27:
If you prefer you can choose gold material for terminals, instead of just copper.

diegoherranz, 2018-10-27:
>> now they are white
They are black, I think. The white board is an adapter board.

Misca1234, 2018-10-27:
Yes, gold would give better "look", but I personally don't mind to use the correct material, then we can easily change the apperance later on in the 3D model viewer.
if we make copper now as gold, we can never "fix it later"

> They are black, I think. The white board is an adapter board.

Well then the colour change even more over time

Cylindric, 2018-10-27:
This is what all current retail ESP32-WROOM-32 modules look like.

[image]

Misca1234, 2018-12-08:
@Cylindric 
Hmm, what happen after this, did you change the colours of the 3D model?
Can I close this thread?

Cylindric, 2018-12-10:
I did not. It wasn't my model, and I'm not familiar enough with FreeCAD to feel comfortable fiddling with them in there.

Misca1234, 2018-12-10:
Ok, then we let them be blue, if I got the time I can change them to black and copper to gold, I close this thread

closed 2018-12-10 by Misca1234