KiCad/kicad-packages3D #423: PCB Terminal blocks phoenix

ferdymercury, 2018-10-22:
I would suggest to add these 3D models

https://www.phoenixcontact.com/online/portal/us?uri=pxc-oc-itemdetail:pid=1725656&library=usen&tab=1
https://www.phoenixcontact.com/online/portal/us?uri=pxc-oc-itemdetail:pid=1725711&library=usen&tab=1
https://www.phoenixcontact.com/online/portal/us?uri=pxc-oc-itemdetail:pid=1725737&library=usen&tab=1

to the already existing footprints:
TerminalBlock_Phoenix:TerminalBlock_Phoenix_MPT-0,5-2-2.54_1x02_P2.54mm_Horizontal
TerminalBlock_Phoenix:TerminalBlock_Phoenix_MPT-0,5-8-2.54_1x08_P2.54mm_Horizontal
TerminalBlock_Phoenix:TerminalBlock_Phoenix_MPT-0,5-10-2.54_1x10_P2.54mm_Horizontal

I have calculated the 3D model offsets of the step files, see attached screenshots:

[mpt.zip](https://github.com/KiCad/kicad-packages3D/files/2502658/mpt.zip)

ferdymercury, 2018-10-27:
Done T2 https://github.com/KiCad/kicad-packages3D/pull/433

ferdymercury, 2018-10-28:
Done T8 https://github.com/KiCad/kicad-packages3D/pull/434

ferdymercury, 2018-10-28:
Done T10 https://github.com/KiCad/kicad-packages3D/pull/435

easyw, 2018-10-28:
Hi @ferdymercury,
you are adding models using original step files coming from manufacturer's sites. This is explicitly NOT allowed. @Misca1234 these models shouldn't have been merged.

Misca1234, 2018-10-28:
Ahhh, I thought they was hand made, sorry, I remove them

Misca1234, 2018-10-28:
@easyw 
Please merge
https://github.com/KiCad/kicad-packages3D/pull/436

Misca1234, 2018-10-28:
@ferdymercury 
I have already provided the removal pushes, you don't need to re open it

closed 2018-10-29 by ferdymercury