KiCad/kicad-packages3D #432: Automatic massive addition of 3D models

Misca1234, 2018-10-27:
Hi

I would like to use several 3D model scripts from https://github.com/easyw/kicad-3d-models-in-freecad such as 
Package_BGA/main_generator.py
GW_QFP_SOIC_SSOP_TSSOP_SOT/main_generator.py

in combination with my script which scans the foot prints, finds the missing 3D models, and extract the dimension E, D and e, either from the file name or the foot print file its self.

The script will default set the parameters A and A1 (height and board separation)
This will create some 100 or more models if successful.

Will we be willing to accept this, that is, merge that push without reviewing every 3D model ?

easyw, 2018-10-28:
@Misca1234  I think this would be nice. 
It would be a good option if your script could generate the parameters file to then build the models, so this generated file could be later modified to eventually adjust some values...
To check all the models in one view, you can load all the footprint in a pcb file and then display the result in pcbnew...
I made a script that can put a footprint list into a pcb file, then you can spread the fp automatically and check it in the 3d viewer 
https://bugs.launchpad.net/kicad/+bug/1766550
https://github.com/easyw/kicad-3d-models-in-freecad/files/1939774/create-pcb-from-modules.zip

Misca1234, 2018-11-06:
>To check all the models in one view, you can load all the footprint in a pcb file and then display the result in pcbnew...

Perhaps the python script should create a pcb file directly as your .bat file does but I can not find the "spreading function" in pcbnew editor ("active foot print mode" is no longer available.
But perhaps the python script could create a .pcb_new file directly with some default spreading

Is it possible to get 
https://github.com/KiCad/kicad-packages3D/pull/448
https://github.com/KiCad/kicad-packages3D/pull/428
https://github.com/KiCad/kicad-packages3D/pull/426
https://github.com/easyw/kicad-3d-models-in-freecad/pull/220

I need them to add the script for the BGA models, make some updates to the SOIC script and generate the last bunch of SOIC models (move them from excludelist to speciallist)

easyw, 2018-11-06:
The spreading function should be still available
https://forum.kicad.info/t/automatically-place-all-footprints-missing-in-kicad5/10534

easyw, 2018-11-07:
@Misca1234
in kicad 5.0.1 the spreading feature is still available. If you have not access to this version, just send me the PCB file with the fp piled and I will return it with the fp spreaded.

Misca1234, 2018-11-07:
Ok, I will do a test shoot to create a pcb file first (I know the size of the foot prints so I will try to place them 10 mm  apart) if it will be to complicated I send it to you

easyw, 2018-11-07:
@Misca1234
with the script a PCB with all fp piled will be auto generated.... You don't need to work manually.

Misca1234, 2018-11-08:
I have not done anything above for the last BGA push instead I did some manual inspection.
The bat file, how do the "feeding file" look like ?

easyw, 2018-11-08:
@Misca1234 
just copy the script in the directory where all the fp are located and launch the script... it will create a kicad_pcb but with all the fp at the same position.
Then the Global Spread Footprints will do the rest of the job
reference discussion [here](https://github.com/easyw/kicad-3d-models-in-freecad/pull/150#issuecomment-383702194)

Misca1234, 2018-11-08:
Hmm, then I should add to the scripts to create a directory with the "affected" foot prints

Misca1234, 2018-11-09:
This project is considered finished, closing

closed 2018-11-09 by Misca1234