KiCad/kicad-packages3D #443: Library package names discrepancy

omogenot, 2018-11-02:
In the new 5.01 version of KiCAD, there is a discrepancy between library component stored package names and actual footprint names.
I.e. Housings_QFN.... vs. Package_QFN
This forces to re-associate all default packages names.

evanshultz, 2018-11-02:
I thought all of those should be fixed but there could be some cruft left over. Thanks for noticing this and reporting it. Can you list the footprints (I assume footprints and not symbols) with the error?

omogenot, 2018-11-03:
Hi,
Hum... Making a list could be long... I think that you could probably write a python program to check symbol footprints vs. actual footprints in Kicad.
From the top of my mind, the Interface_Expansion:MCP23S17_ML is defined with footprint Package_DFN_QFN:QFN-28-1EP_6x6mm_P0.65mm_EP4.25x4.25mm whereas the existing footprint in Kicad library is Housings_DFN_QFN:QFN-28-1EP_6x6mm_Pitch0.65mm.

evanshultz, 2018-11-03:
I asked because I think the problem is your libraries. `Package_...` is the current name in our libraries. You should update all of your libraries and that will resolve the issue.

omogenot, 2018-11-03:
OK. Ready to test. I use Kicad 5.01 on OSX. How do I update the libraries? By the way which library: symbols or footprints ? Isn't the symbol installed with the installation package? Can't it be then a discrepancy between library versions in the installation package for OSX?

evanshultz, 2018-11-05:
KiCad v5 requires you to manually grab the libraries and update them. See the info at http://kicad-pcb.org/libraries/download/. Update all three to be sure.

omogenot, 2018-11-05:
Yes and no... On the download page one can read:

> kicad-unified is the default package that most people should install. kicad-unified contains everything: the KiCad suite, documentation, schematic symbols, footprints, translations, templates, demos, and 3D models

And this is exactly what I did. Therefore, there might be a discrepancy between the libraries included in the installation package and the current ones.
I'll try update the libs.

omogenot, 2018-11-05:
In fact, the "Housing_..." name comes from the ~/Preferences/kicad/fp-lib-table which has not been updated during the installation process.
I deleted this file and it has been re-created with the correct names in it.
May be a step to be added in the installation script.

evanshultz, 2018-11-05:
As I posted in another of your issues, using KiCad and wishlist/bugs are not for this repo. You will not reach the people you want to reach.

So this issue is closed?

omogenot, 2018-11-05:
You can close it. However, can't you pass it along?

evanshultz, 2018-11-05:
Thanks for confirming. I will close.

You're asking me to speak for you and tackle the overhead of this issue. That's your responsibility. The dev team (separate from the libraries) should hear your voice and perspective.

closed 2018-11-05 by evanshultz

omogenot, 2018-11-05:
OK. Thanks for your help. I will make sure to never contribute again.
I close all issues I have opened so far.

antoniovazquezblanco, 2018-11-05:
Please, don't take this personally. Your input is useful but it was not reported in the right place. Don't give up on the program but remember that people here is investing their own personal time for free in order to improve the tool. People here expect users to contribute with their time and effort if they wan't particular things to be implemented.

Thank you for your patience.

evanshultz, 2018-11-05:
@omogenot 
I feel very strongly that the library updating system is not meeting many users' desires, which I believe aligns with your comments above, and I already brought up this issue. The dev team will want to know if many users (like yourself) request the same features. That is why I asked you to get in touch with them and wrote my response to let you know the developers are not watching this repo but they need to hear directly from you because it's what you want that matters.

omogenot, 2018-11-05:
To end up the story, please consider that you are sometimes talking to simple users, not software engineers, and these simple, stupid, users can get easily lost between all these github/Ubuntu/launchpad/whatever in order to find the right place with the right format and the right credentials...
Sorry for not being patient enough (nothing personal as well).

poeschlr, 2018-11-05:
The problem you expirience is documented in great detail here: https://forum.kicad.info/t/i-had-kicad-4-installed-previosly-now-i-updated-to-v5-now-i-have-some-problems-with-the-library-setup/11932

TlDr: The reason you have trouble is because the kicad installers can not update user config files. Which footprints you use is setup in a user setting file. This came up on the mailing list as well. The conclusion has been that there is no easy fix that could be implemented in time for the v5 release.

omogenot, 2018-11-05:
OK then. 
Can't it be that this list has two parts:
* kind of "ROM" fixed part that comes from Kicad as a system default
* Extension made by the user
Therefore, the fp-lib-table file will only contain the user extended part, whereas the fixed part will be automatically pre-pended by the Kicad program itself and not modifiable ?
This will allow you to change the "default" library names at will without breaking the continuity across versions.