KiCad/kicad-packages3D #464: PowerIntegrations_SO-8C does not fit the footprint

antoniovazquezblanco, 2018-11-12:
@Misca1234 A model involved in one of the massive pushes is wrong.

I have found that `PowerIntegrations_SO-8C` does not fit the footprint.

[image]

Misca1234, 2018-11-12:
Jee, I had some trouble with those (PowerIntegrations_eSOP-12B, PowerIntegrations_SO-8, PowerIntegrations_SO-8B, PowerIntegrations_SO-8C')

Let me recheck them, they needed to be hand made

Misca1234, 2018-11-12:
Yes,
It was copy paste error in PowerIntegrations_SO-8, PowerIntegrations_SO-8B, PowerIntegrations_SO-8C from PowerIntegrations_eSOP-12B.
Here is a correction push

3D model push

3D script model push

This one 

[image]

Looks like this now

[image]

3D model push
https://github.com/KiCad/kicad-packages3D/pull/465

3D script model push
https://github.com/easyw/kicad-3d-models-in-freecad/pull/227

closed 2018-11-13 by antoniovazquezblanco