KiCad/kicad-packages3D #467: Problems with TSOP models

markus-k, 2018-11-13:
I noticed two issues with the models added in #445:

Pins on the TSOP-II-54 are way to long:
[image]

The pin spacing on the TSOP-I-32 is wrong:
[image]

Using KiCad 5.0.1 and packages3d from current master (e5adf6).

Misca1234, 2018-11-13:
Fixed, see push
https://github.com/KiCad/kicad-packages3D/pull/470

Misca1234, 2018-11-13:
@markus-k 
Thanks for finding it

antoniovazquezblanco, 2018-11-14:
Please @markus-k verify that everything is working for you :)

closed 2018-11-14 by antoniovazquezblanco

markus-k, 2018-11-16:
Yep, looking good now. Thank you.