KiCad/kicad-packages3D #469: Phoenix PTSM series

tolot27, 2018-11-13:
I started working on script-generated 3D models for Phoenix PTSM connectors. Hence, I raised this issue here to discuss something.

At first, I want to know when to switch from `TerminalBlock_Phoenix.pretty` to `Connector_Phoenix_PTSM.pretty`

BTW: Tomorrow I'll meet Phoenix Contact at the [electronica](https://electronica.de) in Munich. Maybe we can get some support.

tolot27, 2018-11-13:
According to the footprints, I should put the 3D models into `TerminalBlock_Phoenix.3dshapes/TerminalBlock_Phoenix_PTSM...`. But when to separate them like it was done with `Connector_Phoenix_MSTB`?

antoniovazquezblanco, 2018-11-14:
I believe there is not a fixed criteria but there is a rule at http://kicad-pcb.org/libraries/klc/G1.2/ that states that a library must be further split when it contains more than 250 elements.

evanshultz, 2018-11-14:
Stick with separation method used for the footprints for the 3D models. If any changes are needed they can be done to both repos in unison.

The 250 item limit was an arbitrary choice, so it's probably weaker than other areas of KLC.

@poeschlr @SchrodingersGat 
How did or do you demarcate TerminalBlock versus Connector? This was never clear to me.

myfreescalewebpage, 2019-03-15:
@tolot27 is the question still valid ? Do the previous comments answer your issue ?

tolot27, 2019-03-15:
Questions are answered and I almost completed the design of footprint and 3D model. Still have to create a PR.

myfreescalewebpage, 2019-03-15:
I'm closing the issue. Your contribution will be welcome :)

closed 2019-03-15 by myfreescalewebpage