KiCad/kicad-packages3D #471: Regular leds wrong Z placement.

leoheck, 2018-11-14:
I am seeing this for a long time ago.

Footprint: `LED_THT:LED_D5.0mm`
3d model: `${KISYS3DMOD}/LED_THT.3dshapes/LED_D5.0mm.wrl`

[image]

jkriege2, 2018-11-14:
Why is this a bug? Basically you can mount these with any offset from the PCB. Some LEDs even have small nothes on the pins that prevent them from moving all to the board, if the drill hole is not very large: 
[image]

I think if you need a specific distance from the boards it#s better to make your own special footprint, as many people will require many different offsets ...
Remember: this is a very generic footprint and cannot (by definition) fit to all use cases (e.g. we also don't have all colors of LEDs).

Best,
JAN

PS: @evanshultz @Misca1234 @antoniovazquezblanco What do you think? Should we move the LEDs directory onto the PCB, or make a series of different offsets? Maybe we should at least specify the ofset (but then measured from where to where?)

Misca1234, 2018-11-14:
At most we could move them to the board but I do not think we should create a different series of distances.

I would like to vote to not move them, just keep them as they are, but if you all want to move them, it should not be a big problem

evanshultz, 2018-11-14:
I think Jan is correct that this is preference or expectation and not a bug. However, moving them to the PCB is the obvious location to me as well and if that is the place most users expect it's probably the best place to locate these 3D models there to best meet expectations and avoid questions.

I agree the official library should be satisfactory with just one offset.

leoheck, 2018-11-15:
Hm interesting, I believe that by default it should be siting on the board. The lowest it can go is on top of the board. Moving it up is a user configuration.