KiCad/kicad-packages3D #513: Adding a standard length for cutted pins on THT components

Shackmeister, 2019-01-19:
Hi I'm noticing a lot of THT components have different pin length, typically from 2-3mm up to the full length.
I personally believe the pins should have a length equivalent to what ever they would have after mounting. And it's pretty rare to have 3cm legs sticking out underneath your board.

Should we perhaps come up with a standard length for uniformity?

SchrodingersGat, 2019-04-07:
And based on a "standard" 1.6mm board, I'm assuming

antoniovazquezblanco, 2019-05-10:
@poeschlr I don't know if you have missed this one. Should this be moved to the website repo for KLC documentation?

poeschlr, 2019-05-10:
If we document details like this in the KLC then we will end up with a 1000 page novel that nobody is willing to read. Some things simply will have to rely on common sense. To be honest i think nobody really cares about any exact measurement here (mainly because it can not be relied upon anyways.) So anything that looks ok is good enough. (in other words anything between 2 to 5mm would be ok for me. And this really is only a topic for parts that do come in "flexible" leads like a resistor not for connectors, IC packages, ... as these later ones are not expected to be cut.)

antoniovazquezblanco, 2019-05-10:
Perfect!

Do we agree on on a 2 to 5mm non written "standard"? Is the question resolved?

poeschlr, 2019-05-10:
I see it as resolved but would even remove these numbers. As long as a model does not look wrong when it comes to this parameter should it be seen as correct. (no need to take out a measuring tool, your eyes should suffice.)

SchrodingersGat, 2019-05-10:
Agreed, I think common sense is sufficient here.

closed 2019-05-16 by antoniovazquezblanco