KiCad/kicad-packages3D #525: Using 3d models

gcormier, 2019-02-09:
I contribute to the kicad libraries. I have KiCad using the symbols and footprint folders that I sync with Git. I generally have them on the master branch, and I add my own branch when contributing, and revert back to master.

Any thoughts on the best way to have custom 3D models (that don't meet license requirements) in my projects? If I download them and add them to a footprint, it would then cause issues when trying to contribute as footprints would have various 3D models associated with them.

I could copy the footprint to a custom library and do the work, but that would become rather annoying quickly I think

Failing a solution, is there anything planned for v6 that would address this use case?

closed 2019-02-12 by gcormier

evanshultz, 2019-02-12:
You should post these kinds of questions at https://forum.kicad.info/. Wishlist items and bug reports for KiCad should be directed to https://bugs.launchpad.net/kicad.

One idea would be to have a local sym-lib-table and fp-lib-table that are merged with the global ones and project ones. That would allow using a main library (either the KiCad one, a company one, etc.) along with your own personal library for all projects. And then you could continue to have project-level libraries as well.

gcormier, 2019-02-12:
Yup I just brought it over to the forums.