KiCad/kicad-packages3D #547: Add Rotary Encoder 3D Shape

horfee, 2019-05-09:
Hi,

I designed a 3D shape for Rotary_Encoder:RotaryEncoder_Alps_EC11E-Switch_Vertical_H20mm.
Could you add it to the library please ?

[Rotary Encoder.step.zip](https://github.com/KiCad/kicad-packages3D/files/3162875/Rotary.Encoder.step.zip)

Thank you

antoniovazquezblanco, 2019-05-09:
Hi @horfee 

We need you to create a pull request in order to accept your contribution for you to sign the contribution agreement and to review the contents.

Thank you!

horfee, 2019-05-09:
Ok, how to proceed ?

antoniovazquezblanco, 2019-05-09:
You can have a look at the following instructions http://kicad-pcb.org/libraries/contribute/

poeschlr, 2019-05-10:
https://help.github.com/en/articles/about-pull-requests

Franck78, 2020-09-08:
The step provided is fusionned (useless).
I think this is the term, FreeCAD sees a single design.

Some to close this ?

chschlue, 2020-09-08:
There are several issues with this issue, including it being an issue instead of a pull request, but the model being fused is not one of them.

Still closing as the author apparently lost interest.

closed 2020-09-08 by chschlue