KiCad/kicad-packages3D #550: DSUB-9 Packages are missing

Wetmelon, 2019-05-13:
The correct paths are in the DSUB-9 footprints, but the .wrl's aren't in the repository.

The .wrl files exist for 15, 25, 26, 37, and 44 pin variants but not the 9 for some reason...

poeschlr, 2019-05-13:
@Misca1234 you scripted these https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Connector_Dsub
Did you forget to add the 9 pol variants here? (They are defined in the script)

Misca1234, 2019-05-26:
Let me check

Misca1234, 2019-05-26:
They are in this push, waiting for merge
https://github.com/KiCad/kicad-packages3D/pull/500/files

antoniovazquezblanco, 2019-05-27:
Question solved

closed 2019-05-27 by antoniovazquezblanco

jayeye, 2019-10-24:
They were never added. At least they are not there in 58d73640ebb764637eb7bba6290815b84a24b8ad

Misca1234, 2019-11-21:
They are waiting to be merged