KiCad/kicad-packages3D #552: Module Relay_SPDT_Omron-G5LE-1 has 3d link but

BGwiz, 2019-05-17:
Module Relay_SPDT_Omron-G5LE-1 has 3d link but it is not here anymore and thus not included in the current 5.1.2 release.  It looks like it once was here, yes?...
${KISYS3DMOD}/Relay_THT.3dshapes/Relay_SPDT_Omron-G5LE-1.wrl

evanshultz, 2019-05-17:
All footprints include the 3D link so that if/when the 3D models are generated the footprints don't need updated.

These 3D models likely were never created and there is no issue then. If you would like to submit them in accordance with the library policy (not downloads from the vendors) that would be great. Scripting can help to easily make many 3D models and many of them are scripted already.

I'll leave this open a while in case you want to respond but I will close this issue if there's no response.

BGwiz, 2019-05-17:
Ah- I see.  I figured the link meant the file was there at one point and was lost somehow.  I was hoping I was helping.  Thanks for the reply!

myfreescalewebpage, 2019-05-17:
Seems the answer is great for this issue, closing.

closed 2019-05-17 by myfreescalewebpage