KiCad/kicad-packages3D #600: IRM-03-xx_THT model rotated 90 degrees from footprint

LambertusIJsselstein, 2019-07-30:
The 3D model for the IRM-03-xx-THT part in package Converter_ACDC is rotated 90 degrees from the footprint:
[image]

EeliK, 2019-08-28:
The whole model is wrong, it's mirrored. See https://www.tme.eu/Document/4ddeead22020b5eb532ee6fb8f44e788/IRM-03-spec.pdf

Franck78, 2020-09-03:
corrected footprint submitted here:

https://github.com/KiCad/kicad-footprints/pull/2433/commits/4a7a688072b14c1475d704330f5ac3b8a9b280d0

Sorry for the the mess but this whole  github + git is  pile a crap. I wanted 'one or two patches' per branches one branch one PR.

evanshultz, 2020-09-03:
@Franck78 
It looks like the footprint is fine to me. The origin and rotation are correct for a THT footprint. It's the 3D model that needs to be fixed.

Franck78, 2020-09-03:
euh no ! You have the pdf. for comparison.

Franck78, 2020-09-03:
eu yes ! bottom view says the datasheet.
So just make the orientation same as other IRM THT is ok.

And redraw the the 3d

Franck78, 2020-09-04:
@evanshultz ,
regenerated wrl for this thing is here:
Just from the script

https://github.com/KiCad/kicad-packages3D/compare/master...Franck78:pwled?expand=1

evanshultz, 2020-09-04:
@Franck78 
Thanks for looking at this. If you've fixed it, please submit the STEP and WRL files as a PR to the repo so it can be reviewed and merged. And if you corrected a script, please submit a PR for that as well. Again, I'm not too familiar with the 3D model part of the library.

Franck78, 2020-09-04:
* the existing .step is OK. No need to change
* the script is not modified
* the PR done 'pwled', updated with photos

Franck78, 2020-09-05:
while working with the sourcefiles for building this image, I also noticed missing or wrong 3dmodels for
* hilink_hlk
* IRM-05
* IRM-10
* IRM-60
* RAC-05
* TMG-15
* Syntec
*  CRE1
* MGJ2D
* MGJ3
* NCS1S
* TDN-05
* THD-15
* TMR 1SM
* TMR-2
* XPPOWER

Probably a lot of them doable with the script, meaning doing the job of updating the 3drepo only 'once'.

Don't merge to soon, but not in ten years ;)

Franck78, 2020-09-08:
~140 3dmodels ready (scripted).

what next Librarians ?