KiCad/kicad-packages3D #627: 3D model/shape for 1210 SMD Diode missing

StreuB1, 2019-12-19:
Currently doing my first project with KiCad and I am using a 1210 diode but noticed that in my 3D view, the diode was not shown. Navigated to the 3D shapes directory and noticed that the wrl or STEP files for it were missing.

Anyone else notice this? Anyone have the files or a means to create them? I work in SW daily so I can model them but I am not sure if there are any specifics that I need to be aware of.

Thanks!

Brian

poeschlr, 2019-12-19:
Yes we have many footprints without 3d models. this is known and not an issue. The footprints are setup such that a user (or we) can easily add the model without modifying the footprint.

closed 2019-12-19 by poeschlr

evanshultz, 2019-12-19:
@StreuB1 
I believe the generator for these 3D models is at https://github.com/easyw/kicad-3d-models-in-freecad if you want to contribute the missing model or any others.