KiCad/kicad-packages3D #636: Generated VRML do not respects specs

tuarrep, 2020-01-03:
As of http://gun.teipir.gr/VRML-amgem/spec/part1/concepts.html#SyntaxBasics a node name cannot contains minus
> Field, event, prototype, and node names must not begin with a digit (0x30-0x39) but may otherwise contain any characters except for non-printable ASCII characters (0x0-0x20), double or single quotes (0x22: ", 0x27: '), sharp (0x23: #), plus (0x2b: +), comma (0x2c: ,), minus (0x2d: -), period (0x2e: .), square brackets (0x5b, 0x5d: []), backslash (0x5c: \) or curly braces (0x7b, 0x7d: {}). Characters in names are as specified in ISO 10646, and are encoded using UTF-8. VRML is case-sensitive; "Sphere" is different from "sphere" and "BEGIN" is different from "begin."

Lot's of files are generated with minus in identifiers in appearance definitions.
They can be replaced by underscore according to spec

Franck78, 2020-09-08:
https://www.web3d.org/documents/specifications/14772/V2.0/part1/concepts.html#4.1.1
says

> Field, event, PROTO, EXTERNPROTO, and node names shall not contain control characters (0x0-0x1f, 0x7f), space (0x20), double or single quotes (0x22: ", 0x27: '), sharp (0x23: #), comma (0x2c: ,), period (0x2e: .), brackets (0x5b, 0x5d: []), backslash (0x5c: \) or braces (0x7b, 0x7d: {}). **Further, their first character shall not be** a digit (0x30-0x39), plus (0x2b: +), or minus (0x2d: -) character. Otherwise, names may contain any ISO 10646 character encoded using UTF-8. VRML is case-sensitive; "Sphere" is different from "sphere" and "BEGIN" is different from "begin."

So, nothing here, can someone close this....

tuarrep, 2020-09-11:
I'm using threejs to display 3d models.
Kicad's models don't load because of the dash.
They also say the dash isn't allowed by the specs.

See here for reference https://github.com/mrdoob/three.js/pull/18298#issuecomment-570632137

tuarrep, 2020-09-11:
ThreeJs agreed on the change based on the version @Franck78 pointed out that seems to be more recent.

closed 2020-09-11 by tuarrep