KiCad/kicad-packages3D #638: Did I just pus a merge to the master without reviewing?

Misca1234, 2020-01-10:
I get the feeling that I pushed a change directly into the  KiCad/kicad-packages3D  master branch, can someone confirm this?

It is a number of 3d Models for the 15 pol THT dsub connectors.

chschlue, 2020-01-10:
I don't share that feeling.

```
commit c2b92a411adc93ddeeed74b36b542e1057f81a2a (HEAD -> master, upstream/master, origin/master, origin/HEAD)
Author: Stefan <misca@hotmail.com>
Date:   Mon Dec 30 01:48:37 2019 +0100

    Added 3D models of 62-pol dsub connector (#506)
    
    * Added 3D models of 62-pol dsub connector
    
    * Fused models

commit e2682dd685206014c4cfcf806c3d4dbf1cb13e92
Author: Stefan <misca@hotmail.com>
Date:   Mon Dec 30 01:44:45 2019 +0100

    Added 3D models for 9-pol dsub (#500)
    
    * Added 3D models for 9-pol dsub
    
    * Changed the internal step name and removed phantom dot
    
    * Fused models

chschlue, 2020-01-11:
I think this can be closed.

closed 2020-01-11 by chschlue